UnchartedBull/OctoDash #866: Web Interface for tablet control

mmoraes80, 2020-07-31:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
It would be nice to have the same nice interface when accessing Octoprint via a web browser (similar to TouchUI plugin) for when a tablet is being used to access / control Octoprint.

**Link other projects**
None

**Additional context**
None

UnchartedBull, 2020-07-31:
I guess your best shot here probably is the awesome plugin from @StefanCohen which is based on OctoDash. It is not 100% similar, but rather optimized for more interaction. https://github.com/StefanCohen/OctoPrint-Dashboard. I currently do not have any plans to port OctoDash to the main OctoPrint interface. Please close the issue if your fine with the dashboard, otherwise I'll open this for discussion :)

mmoraes80, 2020-07-31:
> I guess your best shot here probably is the awesome plugin from @StefanCohen which is based on OctoDash. It is not 100% similar, but rather optimized for more interaction. [...]

I have using OctoPrint -Dashboard plugin for a while and it is really good. The reason I am asking is that I typically don't stand by the printer at all times (it works in the basement) and I use my tablet to monitor it wherever I am in the house and your UI is undeniable better looking and easier to use on a tablet. 

I can close the topic but I believe this would provide some nice discussion for future implementation.

UnchartedBull, 2020-08-05:
What OS is your Tablet running? If it is running Windows it should be fairly easy to build and .exe file

mmoraes80, 2020-08-05:
> What OS is your Tablet running? If it is running Windows it should be fairly easy to build and .exe file

Android

UnchartedBull, 2020-08-10:
That's not optimal. While electron builder (the tool I use to package the whole thing) does have an android target it doesn't seem to work at all. A few versions have an android installable attached, though no one seemed to be able to install them successfully. I don't have an Android device so there is no need to test this for me really. 

Do you have a technical background and could try working on that? I'm happy to help to get you started, but as I said - I don't have any android devices so I can't help that much after that.

mmoraes80, 2020-08-10:
The Android device is just Chrome web browser... no need to develop anything Android specifically. The idea would be having a customized Octoprint interface to make it look like Octodash. The dashboard plug-in is nice but it has a lot of issues.

redeye86, 2020-08-13:
Shouldn't it be possible to run the development server (npm run start) on the pi and then connect to the server using the browser on the device? I mean you need to persuade the server to accept connections from the network (or setup a reverse proxy) and write a startup script, but this should be doable.

UnchartedBull, 2020-08-18:
Kind of possible, but it would require some changes. I removed the website support, since I'm relying on a few features that are exclusive to electron (i.e. config management). It should be fairly easy to make a website only version - every occurence of `this.ipc` in the app would need to be replaced with proper handling for a website, this could either be ignoring it just all together (update functionality isn't needed from the website) or loading a static file.

jappyjan, 2021-03-04:
i will try my best to create a PR / Fork for this.
Would be kinda nice to be able to have a glance on octoprint via the octodash ui from a phone or browser as it is a really nice ui.

already generated a mac app from the source code as it is way nicer than needing to open the rather slow and ugly octoprint ui in a browser :D

jappyjan, 2021-03-04:
guess i might be able to "mock" the rpc with a web worker script... will see

jappyjan, 2021-03-05:
So maybe someone here with more knowledge on angular might help me out already.
i polyfilled a lot of the electron functions/communication already and now i got a bug where everything is fine on chrome desktop but on ios safari i cannot change the value of the input fields for host and port in the setup screen. (tested on ipad)

to test this on your own machine you can checkout my fork at https://github.com/jappyjan/OctoDash
and run the web version via "npm run start:web", you can then access OctoDash via any browser using the ip of the machine running the code and port 4200

UnchartedBull, 2021-07-28:
Due to more and more electron magic being used I'm closing this here. If someone wants to take this over, feel free to comment and we can re-open this :)