UnchartedBull/OctoDash #869: issue with upgrade to 2.0.0

dmberrycom, 2020-07-31:
**What doesn't work?**
OctoDash 1.5 to 2.0 upgrade via the update script 

wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/update.sh | bash

Error displayed :

Your config is invalid
filament should have required property useM600
octodash should have required property preferpreviewwhileprinting

**What did you already try?**
Looking at sample configuration on github to see if these were declared but appears not.

**General Information:**
Pi 4 , 4gb 800x480 touchscreen (manual install not octopi)
 Raspbian GNU/Linux 10 (buster)
 - OctoDash Version 2.0.0
 - OctoPrint Version [e.g. v1.4.0]

**Additional context**
Add any other context or screenshots about the feature request here.

NickRno77, 2020-07-31:
before I rip into installing V2.0 have you tried installing using the manual process? https://github.com/UnchartedBull/OctoDash/wiki/Installation

Is there a DLP free version of OctoDash 2.0 like there is for V1.5 ?

UnchartedBull, 2020-07-31:
Sorry missed that somehow. I updated the v2.0.0 release, please reinstall and report back if everything is working now. Sorry for the inconvenience!

@NickRno77 Currently not no. I can release a version with DLP disabled once I'm back from vacation in a few days (great idea to release before leaving I know :D). So please stay with 1.5.0 for a couple more days. I'll post the new build in the original issue #584.

NickRno77, 2020-07-31:
@UnchartedBull I'll run with V2.0 with DLP and see how it rolls, I'm using ArcWelder which has solved some of the issues I was having with stuttering.

dmberrycom, 2020-08-01:
@UnchartedBull no need to apologise, just wanted to flag it encase it was an issue, great work on this application. 

A reinstall of octodash via the script fixed the issue, many thanks

Kawuezel, 2020-08-01:
@UnchartedBull Thanks for publishing a version without DLP. I have no more trust in that plugin ;-) Wish you a nice vacation 👍

closed 2020-08-03 by UnchartedBull

NickRno77, 2020-08-17:
> Sorry missed that somehow. I updated the v2.0.0 release, please reinstall and report back if everything is working now. Sorry for the inconvenience!
>
> @NickRno77 Currently not no. I can release a [...]

I was just wondering if you managed to do a DLP free build of V2, I'm rolling with this version ok so far and haven't experienced many problems as I'm using ArcWelder to stop the stutter. Someone else wants to run OctoDash without DLP, I pointed them to the download for the previous build but it looks like it's not there now. Would it be possible to make it available again or do a V2 build free of DLP.  

Many Thanks 

Nick

UnchartedBull, 2020-08-19:
Just now posted it over in the other issue. Here you go: https://drive.google.com/file/d/1-OcUZOLr9zrxfd5P9Ybi81zIekI15GaW

NickRno77, 2020-08-19:
> Just now posted it over in the other issue. Here you go: https://drive.google.com/file/d/1-OcUZOLr9zrxfd5P9Ybi81zIekI15GaW

Thanks for downloading, not sure whats going on but I'm getting following errors.:-

pi@octopi:~/Downloads $ wget -O octodash.deb octodash_2.0.0_armv7l.deb
--2020-08-19 19:21:54--  http://octodash_2.0.0_armv7l.deb/
Resolving octodash_2.0.0_armv7l.deb (octodash_2.0.0_armv7l.deb)... 92.242.132.16
Connecting to octodash_2.0.0_armv7l.deb (octodash_2.0.0_armv7l.deb)|92.242.132.16|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘octodash.deb’

octodash.deb            [ <=>                ]     890  --.-KB/s    in 0s

2020-08-19 19:21:54 (15.6 MB/s) - ‘octodash.deb’ saved [890]

--------------------------------------------------------

pi@octopi:~/Downloads $ sudo dpkg -i octodash.deb
[sudo] password for pi:
dpkg-deb: error: 'octodash.deb' is not a Debian format archive
dpkg: error processing archive octodash.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 octodash.deb