UnchartedBull/OctoDash #873: Thumbnail issue in print screen after V2.0 update

mmoraes80, 2020-08-01:
**Describe the bug**
When I start a print job sliced with Prusa Slicer, a tiny thumbnail is shown inside the progress circle (not even in the center of the circle). Is Octodash supposed to show the thumbnail, the circle, or both?

**To Reproduce**
Steps to reproduce the behavior:
1. Slice a model within Prusa Slicer
2. Send to Octoprint and start printing
3. Octodash main printing screen
4. See error

**Expected behavior**
In the Octodash GitHub main page, there is a picture of the main printing screen with the thumbnail without the progress circle. Has it changed in the new update?

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

thegarbz, 2020-08-01:
The picture on the main page is ancient and predates the progress meter. Ever since support for Prusaslicer was added the preview was always inside the progress circle. 

Also the picture is largest if the dimensions are square. I suggest configuring prusaslicer to make 256x256 thumbnails.

mmoraes80, 2020-08-01:
> The picture on the main page is ancient and predates the progress meter. Ever since support for Prusaslicer was added the preview was always inside the progress circle.
> 
> Also the picture is largest if the dimensions are square. I suggest configuring prusaslicer to make 256x256 thumbnails.

Thank you for the feedback. I was able to increase the size of the Thumbnail to 256x256 (and 512x512 as a test) but the image is still not centered inside the progress circle gauge.

thegarbz, 2020-08-01:
Can't help much beyond suggesting that you very very carefully check your UI and compare it with other examples. I had an issue where UI elements didn't line up correctly which I narrowed down to my display being at the wrong resolution. It stretched to the correct resolution but reported the wrong resolution causing the interface to be offset. I had that on my 3.5" LCD. No further issues with a larger LCD.

For comparison here's a screenshot with 256x256 thumbnails running the v2.0 update:
https://imgur.com/a/pd3LYfF

mmoraes80, 2020-08-02:
> Can't help much beyond suggesting that you very very carefully check your UI and compare it with other examples. I had an issue where UI elements didn't line up correctly which I narrowed down to my [...]

Thanks for the feedback! What would be the correct resolution in this case? I am using an UCTronics 3.5 LCD screen. One thing I noticed is that my thumbnail doesn't; show the background though.

thegarbz, 2020-08-02:
I'd love to know how to disable the background, but that's likely a slicer setting.

Correct resolution is 480x320 according to a quick google search. Worth checking in the raspberry pi setup in /boot/config.txt if that is set correctly.

jneilliii, 2020-08-03:
Looking at the difference between the two screenshots it looks like one of you is using UltimakerFormatPackage plugin and one is using PrusaSlicerThumbnails. UFP always output the same 300x300 square thumbnail size, whereas PrusaSlicer will export the image based on the size configured in the printer's ini file.

jneilliii, 2020-08-03:
I'm curious, does the discard button close the thumbnail view and show the print progress?

UnchartedBull, 2020-08-03:
That image you posted is still valid. You can open this screen by loading a file (clicking the blue arrow up button in the file details view). During Print the image will be within the progress ring and smaller. You can also switch back to percentage by just tapping the image once. There are currently some discussion on how to make the image bigger without everything being cluttered.

If your image isn't centered with a square aspect ratio please post a screenshot/photo here.

mmoraes80, 2020-08-03:
I managed to have the screen showing up but I don't the image size that big (proportional to the other items of the screen). Also, during printing, the little thumbnail is aligned to the top (even at 300x300) not centered in the middle of the progress gauge.

ruralaccent, 2020-08-05:
I get the small. The thumbnail from Prusa Slicer is tiny and off centre within the progress 'circle' in Octodash.

UnchartedBull, 2020-08-05:
I'll have a look at this later

PR #938 (2020-08-25): Fix preview image while printing
Closes #873
closed 2020-08-25 by UnchartedBull