UnchartedBull/OctoDash #874: M600 Setup

seacred, 2020-08-01:
**What doesn't work?**
M600

**What did you already try?**
Setting up Marlin 2.x for M600

What exact Setup is needed to make M600 work properly?
Should M600 used directly on the Board as FilRunout connected to an endstop or need i rather use it on hostside over GPIO?

thegarbz, 2020-08-01:
Correct me if I'm wrong but the way I understand this would be an Octoprint issue as the M600 gcode isn't trapped in Octoprint and the code is just streamed as normal Gcode to the printer. 

Marlin after receipt of the M600 sends back "echo:busy: paused for user" but Octoprint does nothing with this and by extension neither does Octodash.

What is not working properly here? Octoprint is still in the process of printing, not paused and the printer is simply waiting on a user input, as soon as you handle the filament change it will resume.

seacred, 2020-08-01:
I'm totally aware of how octoprint handles M600 - since octodash 2.0.0 should support M600 I'm wondering how this is done.
M600 over Marlin via action:command or via plugin through octoprint.

thegarbz, 2020-08-01:
Are you talking about the settings option "Use M600 to change filament"? Because I requested that one and all that does is change the way filament manager works. 
Previously filament manager would Preheat > Retract > Wait for user input > Purge a predefined amount and wait for user input. With that option enabled it simply Preheats > Sends M600 and let the printer handle the rest of the user interaction (which is good since if you jam the filament during changing Marlin gives you the option to disengage the steppers so you can bust out the screwdriver)

That option (if that's what you're referring to) has nothing to do with printing at all.

seacred, 2020-08-01:
So i'm still need a LCD to perform that user interaction?
Hoped octodash can perform that now for Version 2.0.0

thegarbz, 2020-08-01:
On the printer? Yes. M600 is a generic single instruction, the printer has a whole routine which its firmware goes through which varies between different brands of printers. This is why I asked to have the M600 for the filament manager screen option added, because I found on Prusa i3 MK3s it would pointlessly double extrude when Octodash tried to manage the filament change. 

Same in printing. Octodash doesn't know the steps your printer goes through so it can't manage the process at all. All it could conceivably do is not use M600 at all and manually run through gcode to park, retract, purge etc. I believe there's a plugin for Octodash that traps the M600 command and takes over through manual gcode precisely for printers which don't support M600.

closed 2020-08-01 by seacred

UnchartedBull, 2020-08-03:
M600 will be supported in one of the next versions. Support meaning that if OctoPrint pauses OctoDash will also pause. This should allow together with rewriteM600 plugin for at least rudimentary support of M600.