UnchartedBull/OctoDash #875: Prusa MK3s setup values

Prestat88, 2020-08-02:
Hi,

first of all thank you for creating such a nice program! I'm in the middle of setting up Octodash on my Prusa MK3s and need some information which i can't find anywhere else.

It's asking for filament feed length and feed speed. Does anyone know the values for a stock MK3s Extruder?

Thanks in advance!

Kai

thegarbz, 2020-08-02:
Not sure on the values to use normally, but as a MK3s user I opened a feature request to use M600 on filament change. You can tick that in the settings and then the filament length and purge distance can be set to zero. This will cause the software to simply trigger M600 filament change process. 

I suggest this because it allows you to use your printer's firmware features: Autoload filament, confirmation that filament is loaded correctly, disengaging of the motors if the filament jammed etc.

UnchartedBull, 2020-08-03:
Filament feed lenght should be somewhere around 50-60mm and feedspeed should be rather slow (like 10, maybe 20). I would suggest starting with a feed length of about 45mm and slowly increasing it until you can pull out the filament after retracting without any issues. You can also measure the distance from the extruder wheel to the nozzle - that is the feed length :)

Prestat88, 2020-08-04:
I tested 50mm which was not enough. 60mm is working fine. Compared to the standard Prusa procedure the unloading speed is really slow (Prusa is at least 3x faster). Does the feedspeed value affect the unloading speed as well?

When executing the filament change on octodash the Prusa LCD is also showing the unloading procedure. As autoload is active, the procedure continues after inserting new filament (purging filament, asking of color is correct y/n, ....). It seems to overwrite the procedure which was started by octodash. After that i set the use M600 for filament change on the octodash settings. But the Prusa is overwriting this procedure as well. Do i have to dissable autoload in the Prusa LCD? Maybe you can explain me the filament change feature on octodash. Maybe i'm using it wrong because it's much more complicated than simply using the unload command on the Prusa LCD.

Apart from that Octodash is working really fine with the new Hyperpixel 4.0:)

PS: I hope you understand what i mean. English isn't my native language:P

thegarbz, 2020-08-04:
Yes the printer overriding Octodash's procedure is why I requested M600 support.

What happens (or should happen when M600 is active:

1. Select Filament
2. Set temperature
3. Wait for heating...
4. M600 issued. 
a) The printer should park and beep to alert the user to eject. User presses the button on printer and filament is ejected.
b) Printer prompts if Filament was ejected correctly. User presses button to confirm.
c) Printer prompts to insert new filament (this is why I'm confused when you say the printer autoload overrides the M600 process, the printer should be waiting for the user here) User inserts filament and it auto loads. 
d) User confirms that filament loaded correctly. Printer purges. 

5. The entire time when the user interacts with the LCD on the printer, Octodash has no idea so all it's doing is displaying a prompt to load new filament. User presses Done or Skip
6. Octodash shows purging filament screen. (oh if you haven't already, set purge length and feedlength to 0mm in the settings, the printer has already purged, no need to do it again). Users presses Done or Skip

That's it. Filament is loaded. 

Is it more complicated than using the filament unload command on the printer? Yes
Is it more complicated than using the filament unload command on the printer, then logging into the octoprint website, then opening filament manager, and then selecting the filament you just loaded? No. - That's the most important feature here. Not that it's faster or easier than the printer's process, but rather that it automates the Filament manager for tracking how much filament you have left and how much filament and which filament is in use during the print.

UnchartedBull, 2020-08-05:
The feedspeed does change exactly that (loading and unloading) :)

I don't know about the Prusa Autoload feature, since I don't have a Prusa printer. Normally the printer shouldn't interfer with the octodash filament change procedure. Does your printer have a filament sensor btw? 
I guess you could try disabling filament autoload. If you don't use the M600 option you shouldn't need to touch the display of your Prusa during the whole process. So please try it with autoload disabled or use M600 as @thegarbz mentioned.

Prestat88, 2020-08-05:
Thanks thegarbz and UnchartedBull for your responses!

thegarbz: You are absolutely right. I just tried the unloading process using m600 and everything happend as you described. I didn't know about the filament manager plugin and it's features. I just set it up and now i understand the benefit:) I'm really new to 3d printing and RPi coding, so please don't be mad with me if i have silly questions... In 03/2020 i didn't even know how a 3d printer works and what a raspberry pi does:P

UncharteredBull: Yes the MK3s has a filament sensor sitting just over the extruder gears. When i try the unloading without M600 i will let you know (but i guess i won't work correctly with the Prusa or why did thegarbz request the M600 feature?)

Best regards from Germany

Kai

thegarbz, 2020-08-05:
No worries, hit me up with silly questions anytime. Either through here or u/thegarbz on Reddit, or just jump into r/prusa3d they are good for helping too.

Groetjes from across your western border :)

UnchartedBull, 2020-08-05:
The OctoDash procedure should work just fine with a Prusa. The M600 command just offers faster cancelling (unless you enable a feature in Marlin to stop GCode while still processing). It's up to personal choice I guess. I don't like using the integrated LCD that much, but that's what the config is for so everyone can customize to his/her liking :)

So many Europeans here, so greetings from another German :)

closed 2020-08-05 by UnchartedBull