UnchartedBull/OctoDash #880: Filament option don't read my Feed Sheed Setting

zenaro147, 2020-08-04:
**What doesn't work?**
When OctoDash try to remove my filament, it removes the entire filament very slowly (300mm/min or 5mm/sec). I set the Feedspeed to 20mm/s and the FeedspeedSlow to 5mm/s. I tried to use the values in mm/min but still the same.
I can't proceed with the filament change because my Octoprint disconnect my printer by TimeOut. 

I check de Terminal via Octoprint and OctoDash send multiples "G1 E-100 F300" instead one "G1 E-700 F1200".

PS: 

**What did you already try?**
I changed the settings using SSH and directly using Pi

**General Information:**
 - Hardware Raspberry Pi 3 B
 - OS Info: Raspbian Debian 10 buster / OctoPi 0.17.0
 - OctoDash Version 2.0.0
 - OctoPrint Version 1.4.0

**Additional context**
I don't use M600 command, and when I run a code manually to change my filament with the same speed, it done fast.

UnchartedBull, 2020-08-05:
Please check your OctoPrint Profile. You can define max speeds per axis there, which can't be overwritten by OctoDash AFAIK. Let me know if this works:

zenaro147, 2020-08-07:
> Please check your OctoPrint Profile. You can define max speeds per axis there, which can't be overwritten by OctoDash AFAIK. Let me know if this works:

Thanks! Now it's working perfectly! 👍

closed 2020-08-07 by zenaro147