UnchartedBull/OctoDash #882: Ability to operate octodash if printer is off or not attached

TxBillbr, 2020-08-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. If applicable also link other issues / bugs here.
It would be very useful for the "sleeping" screen to allow access to Octodash even if the printer is off. I use printer power-off controls and sleep prevents me from doing anything with the printer unless it can connect to the printer.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like a boolean flag in the settings that allows use of the interface even without the printer on. It might be useful to have a warning message at the top or bottom (or background color change) indicating nothing can be done with the printer until it shows up. My intent would be to add a custom button for "power-on", "shutdown", and "restart" that would allow for local pi  control that doesn't require printer access.

**Link other projects**
If you want OctoDash to include other OctoPrint plugins or similar please link them here.

**Additional context**
Add any other context or screenshots about the feature request here.

Thanks for the great tool. I use it on all my printers.

fenix8k, 2020-08-20:
I can't believe this is not a feature yet or any guide on how to achieve this...

Any way of disabling the sleep function manually?

UnchartedBull, 2020-08-24:
Currently not no, since the API calls need to be shut down / suppressed if OctoPrint isn't connected. Might be something possible once push updates are used, but most likely only the control screen and nothing else.

bepstein111, 2020-09-17:
To be clear, is there currently a way (in normal operation) to shut down/restart the client from OctoDash? Doesn’t appear to be any in the screenshots and I’m hesitant to replace octoscreen (even though I can barely read it) because those are two main functions I use it for.

UnchartedBull, 2020-09-17:
There is a custom action for restarting OctoPrint and the Pi plus shutting down the Pi (OctoPrint can't be shut down via the API): https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions. Does that fit your needs?

TxBillbr, 2020-09-20:
I think so. I will take a look at it this week. Thanks!

closed 2020-09-20 by TxBillbr

bepstein111, 2020-09-21:
> There is a custom action for restarting OctoPrint and the Pi plus shutting down the Pi (OctoPrint can't be shut down via the API): https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions. Does that fit your needs?

Should do. Will try it out shortly. As long as it doesn’t use anything specific to the pi, I’m using a beefier ryzen build for octoprint.