UnchartedBull/OctoDash #884: no unsleep after connect from octoprint

whoim2, 2020-08-06:
have PSU. In 1.5v octodash unsleep automatically after connect to printer (klipper on octopi). In v2 "octodash sleeping" on screen while not touch.

UnchartedBull, 2020-08-18:
Hmmm weird, shouldn't have worked in 1.5. I can't really check whether the printer is connected during sleep, since this would cause an overflow somewhere in OctoPrint. So not possible unfortunately - you need to touch the screen once to wake everything up.

closed 2020-08-18 by UnchartedBull

whoim2, 2020-08-20:
> Hmmm weird, shouldn't have worked in 1.5. I can't really check whether the printer is connected during sleep, since this would cause an overflow somewhere in OctoPrint. So not possible unfortunately - you need to touch the screen once to wake everything up.

It precisely connected itself when connecting, for example, in the octoprint web interface or via a telegram bot. I remember this moment, it was convenient. For fun, I'll try to install the old version.