UnchartedBull/OctoDash #890: 409 conflict error

megatech1966, 2020-08-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**[image]
Steps to reproduce the behavior:
1. Go to files
2. Click on local
3. Select file to print
4. See error 409 conflict. Still prints ok. 

**Expected behavior**
No error message

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Hardware pi 4. CR10S3
 - OS Info Raspbian Buster Lite
 - OctoDash Version [v2]
 - OctoPrint Version [v1.4.2]

**Additional context**

UnchartedBull, 2020-08-10:
Has this started with 1.4.2, so doesn't occur with 1.4.0? I've been away a couple of days, so I'm just catching up.

megatech1966, 2020-08-10:
Not sure. I would have to see if I can downgrade OctoPrint to 1.4.0

megatech1966, 2020-08-14:
Does not do it with OctoPrint 1.4.0

UnchartedBull, 2020-08-18:
Haven't had any time to upgrade my printer to 1.4.2 yet. I'll try to find some time this week.

UnchartedBull, 2020-09-02:
Could you maybe send over that GCode file? I can't reproduce this issue on 1.4.2 on my machine :/.

megatech1966, 2020-09-02:
I can’t get it to do it anymore. But sometimes I do get this error.

megatech1966, 2020-09-02:
I got it to do it .
https://www.dropbox.com/t/XUMcYjkUalIKvYrs

From: Timon G.
Sent: Wednesday, 2 September 2020 5:48 PM
To: UnchartedBull/OctoDash
Cc: megatech1966; Author
Subject: Re: [UnchartedBull/OctoDash] 409 conflict error (#890)

Could you maybe send over that GCode file? I can't reproduce this issue on 1.4.2 on my machine :/.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

UnchartedBull, 2020-09-07:
Thanks. I'll try that and report back :)

UnchartedBull, 2020-09-15:
I've tried multiple things and I'm just not able to replicate this on my end. With one of the next releases the API calls will be replaced by the OctoPrint JS Client, which should hopefully fix this issue. Until then I probably won't be able to provide a fix for this. Sorry for the inconvenience. Will most likely be fixed together with #595

titis14, 2020-09-18:
Hello, I have the same error I think it came from a usb problem between my screen, the printer and the cam but apparently it's something else.
[image]

UnchartedBull, 2020-09-21:
@titis14 the error you're getting is something different. The weird thing is that the values for your printer are all available (temperature / status / ...). What happens if you close the card, does it come back again? Mabye OctoPrint just had a small hickup.

megatech1966, 2020-09-21:
Card goes away. I am going to do a fresh OctoPrint install. I will see if it still happens.

titis14, 2020-09-21:
@UnchartedBull If I type on the notification it goes away and no problem afterwards. I have to do a test without the cam because at the beginning of the installation of octodash I did not have this error.

megatech1966, 2020-09-26:
I did a fresh install of OctoPrint and octodash. Still get errors when picking gcode files. This I didn't install samba. Thought that might fix it. It didn't 🥺

megatech1966, 2020-09-28:
I think I fixed it. I was using the API key which was under the user settings. I changed it to the global API key and haven't seen that error message since.

UnchartedBull, 2020-09-28:
Great! Closing this for now. If the issue reappears please reopen :)

closed 2020-09-28 by UnchartedBull

titis14, 2020-09-28:
Switching on the printer and octodash today I had no error.

megatech1966, 2020-09-29:
I thought I had fix the 409 error when I used the global API key, but no it still happens😢. Also tried port 80 instead of 5000. No change🥺 Tried uninstalling octopod. No change😭 Is there any else that I can try?

megatech1966, 2020-09-29:
> Switching on the printer and octodash today I had no error.

Sometimes it works. Sometimes it doesn't 🥺

megatech1966, 2020-10-01:
It's been solved by James Mackay. For some reason my touchscreen is recording 2 taps when I tap on the print icon too slowly.

titis14, 2020-10-02:
> 
> 
> > Switching on the printer and octodash today I had no error.
> 
> Sometimes it works. Sometimes it doesn't 🥺

The same for me during my last 2 ignition 409 error tap on it and it works anyway.

megatech1966, 2020-10-02:
Did you see me post? I found what was causing it. A dev on Facebook suggested it and he was right.

titis14, 2020-10-04:
> 
> Did you see me post? I found what was causing it. A dev on Facebook suggested it and he was right.
> […](#)
> Sent from my iPhone X 256Gb

I have the error already starting without touching the screen and even without turning on the printer.

megatech1966, 2020-10-04:
You will get all sort of errors if the printer isn’t on.

UnchartedBull, 2020-10-05:
@titis14 could you maybe elaborate on the errors you're getting? Maybe attaching a screenshot. Do they also occur if the printer is turned on?

titis14, 2020-10-05:
> 
> 
> @titis14 could you maybe elaborate on the errors you're getting? Maybe attaching a screenshot. Do they also occur if the printer is turned on?

Hello, I put a screenshot above with the orange screen. I also have the error if the printer and turn on. I don't know what it came from because if I type on the message it goes away and everything works normally.

UnchartedBull, 2020-10-09:
Oh didn't saw that you're the same guy. As stated above please wait for release 2.2 or 2.3, which will replace the API with OctoPrint websocket. This should get rid of the error. Sorry that you need to close this each time until then :/

titis14, 2020-10-10:
> 
> 
> Oh didn't saw that you're the same guy. As stated above please wait for release 2.2 or 2.3, which will replace the API with OctoPrint websocket. This should get rid of the error. Sorry that you need to close this each time until then :/

No problem ;) you do a great job is for free.