UnchartedBull/OctoDash #892: Octodash outputs to HDMI and not GPIO 3.5" TFT

Urzulus, 2020-08-10:
**What doesn't work?**
Had Octodash 1.5 running previously. Upgraded octopi, kept getting rolling 403 errors and decided to fresh install. And this is where the thousand years of pain started.

Octodash outputs to HDMI, 3.5tft displays CLI

**What did you already try?**

- Fresh install via auto script a fair few times. 
- Manual installation, seemed to fail
- Installed drivers before and after Octodash install
- installed adafruit drivers (because of the "would you like the HDMI to mirror" option) It didnt work

**General Information:**
 - Hardware Raspberry Pi 3B+
- Generic 3.5" tft using https://github.com/CytronTechnologies/xpt2046-LCD-Driver-for-Raspberry-Pi Drivers
 - OctoDash Version 2.0.0
 - OctoPrint Version octopi-buster-lite-0.17.0 fresh install

**Additional context**

- Was working prior. so setup is capable of running it
- HDMI outputs octodash, while tft function still works with CLI overlaid on 3.5"
- same setup also works with octotft

Urzulus, 2020-08-10:
Wow this is awkward. 

A non standard driver from waveshare fixes the issue. Only took me 8 hours

closed 2020-08-10 by Urzulus

UnchartedBull, 2020-08-10:
Would you mind linking that driver so I can add it to the troubleshooting guide if other users encounter the same problem?

Urzulus, 2020-08-10:
Sure, hopefully you can help with the next problem I'm currently having.

https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)

kitcarfreak, 2021-01-27:
> Sure, hopefully you can help with the next problem I'm currently having. https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)
> […](#)

Thank you very much for the link, had same issue with a 4 inch display with bogus drivers, used these (LCD4) and thadaa :D happy now