UnchartedBull/OctoDash #896: Bootup screen split into 2 diagonals and display says "no signal".

sr1329, 2020-08-10:
**Describe the bug**
Installed manually Octodash 2.0.0 ARM64 version onto Octopi 18.1 64bit. Followed all directions including autologin into the text interface. Usually an HDMI attached display would show a bunch of scrolling text and then a prompt. Now it shows two weird distorted diagonal lines of text. It resembles how a VCR head writes data diagonally on a rectangular piece of magnetic tape. 

I went to troubleshooting and got while attempting a manual start:

pi@octopi:~ $ octodash
The futex facility returned an unexpected error code.
Aborted
pi@octopi:~ $ export DISPLAY=:0
pi@octopi:~ $ ratpoison
ratpoison: There can be only ONE.
pi@octopi:~ $ octodash
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to open vc4 (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: vc4
MESA-LOADER: failed to open kms_swrast (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: kms_swrast
MESA-LOADER: failed to open swrast (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load swrast driver

**To Reproduce**
1. Boot up machine
2. Attempt to manually start Octodash

**Expected behavior**
Boot into OctoDash UI

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Raspberry Pi 4B
 - Octopi 18.1 Beta
 - OctoDash Version 2.0.0
 - OctoPrint Version 1.4.2

**Additional context**
64 bit

sr1329, 2020-08-11:
So I decided to try again on Octopi 17. No diagnosable weirdness however it too won’t launch into Octodash and gives errors. However it was updated to Kernel 5.4 and I believe that is what is breaking everything. I will go back and reinstall Octopi 17 and neglect to run apt-get upgrade and try again. This is on my test raspberry pi 4B 4GB.

sr1329, 2020-08-11:
So I have tested on a new install with octopi 17. Apart from some issues reaching some mirrors to download packages (where it tells you to apt-get update - never change that to upgrade). 

So it works now and I think the issue is with the new kernel. It has fully installed on the 32 bit updated kernel but doesn’t load. Ratpoison simply cannot open display. 

As for Octopi 18 64bit that is still very weird and I’m not sure what happened there with the diagonal text scrolling.

sr1329, 2020-08-13:
Been playing with it and really like it. I discovered a fix for Hyperpixel 4 for use on kernel 5.4 or even 64bit raspbian. Apparently it works if you fix the hyperpixel driver so Octodash itself is okay on 5.4 or even 64 bit. I will try that install when I get home. 

Regarding HDMI output I suspect it is an issue where it says cannot open Display 0. Hopefully in time it will work on both Hyperpixel and HDMI.

Tobi0892, 2020-08-17:
@sr1329 Thank you for investigating the issue, I can confirm the bug

UnchartedBull, 2020-08-18:
@sr1329 Could you maybe link the fixed driver for Hyperpixel here? If I understand you correctly everything is working once the correct driver has been installed?

@Tobi0892 did you manage to fix the issue as well or are you still having problems?

Tobi0892, 2020-08-18:
@UnchartedBull I hadn't yet the time to investigate the issue further. I'll keep you updated

UnchartedBull, 2020-08-19:
@sr1329 the touch boxes should normally cover the whole icon and / or text. I'll have a look at that, but the resolution shouldn't be a huge issue. Does this happen on other buttons as well or only the back button?

I might redirect a few other issue here, which all seem to have the same problem with Kernel 5.4. There also was a driver linked in this issue: #892. So sorry for all the possible message, but I want to condense this into one issue so others can easily find it.

sr1329, 2020-08-19:
Yes I agree all 5.4 issues would be better consolidated.

The back button is the most evident. It took me a some time to figure out where exactly to touch it. But I feel all buttons have some degree of touch not responding.

In my case it seems towards the bottom I have to touch above the text and towards the top I have to touch slightly below the text.

In general i tend to hunt for the touch sensitive area.

UnchartedBull, 2020-08-19:
[image]

That is the size of the back button (on 1024x600). Could you maybe touch & hold the same button in the files view and check how big your light grey box is?

sr1329, 2020-08-19:
Yes it is the size but the gray box only comes up when pressing the letter “b”

[Screenshot 2020-08-19 at 19 50 32]<https://urldefense.proofpoint.com/v2/url?u=https-3A__user-2Dimages.githubusercontent.com_19818061_90671721-2D5f77b680-2De255-2D11ea-2D97f1-2D9b4e8479c8b1.png&d=DwMCaQ&c=slrrB7dE8n7gBJbeO0g-IQ&r=i0iMjlNrlUgfsH63L7eELQ&m=tE9iMUab4l4_F0pCLIspN-WxjsmRTsV02k7HxVMRepE&s=TQCsO_865F4tkHvzpnwsNz6qMEr8rCEX-xMLdhCdj5c&e=>

sr1329, 2020-08-19:
Seems to be working better now. I think it requires a longer touch than i’m used to. I tried octoscreen for example and it accepts touches very quickly. I have set the touch refresh to 330ms.

sr1329, 2020-08-19:
I think it’s fine. I’m probably not used to nice delay and fade animation.

closed 2020-10-14 by UnchartedBull