UnchartedBull/OctoDash #897: Remove RatPoison banner on startup

TheSin-, 2020-08-10:
in the ~pi/.ratpoisonrc

add

```
startup_message off
```

to remove the ugly startup banner, make it look more like a clean UI.

UnchartedBull, 2020-08-18:
Will add this as an option to the installer script :) Thanks for the tip!

pesawyer, 2020-08-27:
> Will add this as an option to the installer script :) Thanks for the tip!

Recommend setting it to "off" as the default.

teachingtechYT, 2020-10-23:
I'm trying to make this change post installation.
`~pi/.ratpoisonrc` was not present so I created it and added the line, but it has had no effect.
Any tips to remove the message would be appreciated.

UnchartedBull, 2020-10-23:
Just had a look at my system and `~/.ratpoisonrc` wasn't present for me as well. After creating it and just adding the single line @TheSin- posted it worked just fine for me. I guess two things to try here first:

- Is the file owned by `pi:pi`?
- Maybe try changing the permissions (`chmod 600 .ratpoisonrc`), it worked for me with default permissions as well but it may be something to try.

After that `ls -la` should look like this for `.ratpoisonrc`:
[image]

Might also be worth a try to create `/etc/ratpoisonrc` as described here: https://www.nongnu.org/ratpoison/doc/Startup-file.html. Let me know if this works for you, or whether you still have issues :)

Btw, are you running any other Window Managers beside ratpoison?

teachingtechYT, 2020-10-23:
Thanks for the response. Previously I created the file as root so I deleted it and created it as pi which did the trick.

UnchartedBull, 2020-10-23:
Yeah the whole Linux permission/user/group system can be tricky, I also mess that up quite a lot. But great that you figured it out and that it's working now! 👍

closed 2025-03-17 by queengooborg