UnchartedBull/OctoDash #900: What version of Octoprint to use.

dmiart, 2020-08-12:
**What doesn't work?**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Used latest octoprint and apt-get update && apt-get upgrade. Auto install scripts completes fine, but ratpoison and octodash doesn't start. 

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.
Tried fresh image of octopi and followed all troubleshooting steps from wiki. 

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

Raspbian Buster Lite OctoPi 0.17.0
Octodash 2.0.0

**Additional context**
Add any other context or screenshots about the feature request here.

My HDMI device is a regular HDMI 1080p monitor. Maybe that's the problem?

thegarbz, 2020-08-12:
Not an Octoprint issue since Octoprint/Octopi by default doesn't ship with any necessary functions for a GUI. 

Debug steps:
Do you see a console on bootup? > Check the settings in /boot/config.txt / or run raspi-config
Can you log into the console and type startx? 
Can you start a window manager after starting x? 
The last step would obvious be starting octodash and that clearly isn't working or we'd not be here.

dmiart, 2020-08-12:
> Not an Octoprint issue since Octoprint/Octopi by default doesn't ship with any necessary functions for a GUI.
>
> Debug steps:
> Do you see a console on bootup? > Check the settings in /boot/config.txt / [...]

I was able to make it work but after I manually start xinit from console from user pi and then in GUI terminal I type ratpoison & followed by octodash otherwise it just stays on the terminal login.

dmiart, 2020-08-12:
Ok so the problem was the installation script created .xinitrc for root user but was supposed to do it for user pi ? I manually followed the steps and now it starts.

thegarbz, 2020-08-13:
Sounds like you were root instead of the pi user when you ran the script. Good to hear it's working.

dmiart, 2020-08-13:
> 
> 
> Sounds like you were root instead of the pi user when you ran the script. Good to hear it's working.

It's possible that I was, but I also run the installation script the second time as a pi user and it didn't create that script.

Swap-File, 2020-08-14:
If I recall, my fresh install that I made a few days ago went like this:

First I burn OctoPi 0.17 image to an SD Card, and set up the basics.
https://octopi.octoprint.org/latest

I log in, see a login message saying installing desktop support is done via a separate script at my own risk.  I do it anyway.
sudo /home/pi/scripts/install-desktop
(with sudo)

Reboot, then I run the Octodash script:
bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)
(no sudo) 

Reboot again.  Done.

I ran into some oddities if I didn't install "/home/pi/scripts/install-desktop" first.

dmiart, 2020-08-16:
Yeah, looks like I accidentally ran the installation script as root. 

Burned new Octoprint image
Changed HDMI settings in config to match my touchscreen HDMI resolution
Ran Octodash installation script
Let it reboot 
Logged in as pi user on TTY
Octodash launched right away.

All without any additional steps.

closed 2020-08-17 by dmiart

UnchartedBull, 2020-08-18:
Yeah I currently am using the current users home directory, since that was the fastest way. I'll try to improve the whole installation process for one of the next releases. But great that everything is working for you! Hope you enjoy OctoDash :)