UnchartedBull/OctoDash #906: Wrong filament used calculation?

Protoncek, 2020-08-14:
I'm not sure if this is OctoDash's error, but i have "a problem": does anyone of you noticed: "filament used" is different on OctoDash than the one in PrusaSlicer. Example: a small hook, in Slicer says filament used 4,36g, while in OctoDash file info says 13.7g. Where to look for error?

Thanks!

UnchartedBull, 2020-08-18:
That could very well be an OctoDash error. The filament calculation seems to be kind of flaky. I'll need to rewrite that part, will be included in one of the next versions.

TacoScientist, 2020-09-07:
I'm having the same issue. Cura quotes at 19g, but display says 55.9g. Its pretty consistently off by a factor of just below 3.

robwasteground, 2020-09-16:
Same issue here, Cura (and OctoPrint) show what seems to be correct length (5g, 1.72m), OctoDash shows 15.3g. It always seems to be off by a factor of somewhere around ~3 here as well.

UnchartedBull, 2020-10-14:
Ok this was a really weird error. OctoPrint seems to be reporting the wrong filament volume (should be 2.5something in the screenshot), this then offsets OctoDash. Don't really know why the amount is still correct then. I'll revert back to calculate everything on my own from filament length, which should hopefully fix the problem.

[image]

PR #1082 (2020-10-14): Calculate filament weight based on length
Fixes #906.
closed 2020-10-14 by UnchartedBull