UnchartedBull/OctoDash #909: Octodash image as splash

dmiart, 2020-08-15:
**Is your feature request related to a problem? Please describe.**
Replace  /usr/share/plymouth/themes/pix/splash.png with Octodash themed image.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Replace  /usr/share/plymouth/themes/pix/splash.png with Octodash themed image as an option in the installation wizard. 

[image]
[image]

Something like this. 

[image]

UnchartedBull, 2020-08-18:
That would then replace the pi logo during boot, right? Looks cool! Linked to #897 kind of to make a fully OctoDash themed distro.

dmiart, 2020-08-18:
> 
> 
> That would then replace the pi logo during boot, right? Looks cool! Linked to #897 kind of to make a fully OctoDash themed distro.

Yeah, this replaces the welcome to raspbian desktop picture. Also have to enable the splash display in raspi-config.

TheSin-, 2020-08-19:
Don't forget to make sure rpd-plym-splash is installed

TheSin-, 2020-08-19:
Also add

disable_splash=1

to config.txt

and add

logo.nologo vt.global_cursor_default=0

to cmdline.txt

this will make it truly silent

https://scribles.net/silent-boot-up-on-raspbian-stretch/

dmiart, 2020-08-19:
What if you send cls in couple of spots before ratpoison? I can't test on mine as I am printing atm. I suspect the text will still be briefly visible though.

TheSin-, 2020-08-19:
Okay I finally did it based on this link

https://scribles.net/silent-boot-on-raspbian-stretch-in-console-mode/

touch ~/.hushlogin

and

in /etc/systemd/system/getty@tty1.service.d/autologin.conf replace

ExecStart=-/sbin/agetty --autologin pi --noclear %I xterm-256color

with

ExecStart=-/sbin/agetty --skip-login --noclear --noissue --login-options "-f pi" %I $TERM

and my one addition is in .bashrc change

source /home/pi/scripts/welcome
if [ -z "$SSH_CLIENT" ] || [ -z "$SSH_TTY" ]; then
    xinit -- -nocursor
fi

to

if [ -z "$SSH_CLIENT" ] || [ -z "$SSH_TTY" ]; then
    xinit -- -nocursor > /dev/null 2>&1
else
    source /home/pi/scripts/welcome
fi

This produces a few second blank screen while ratpoison loads but zero text at all.

I also had to make the splash screen 1280x800 for my display and I made it all grey without the black but that is just preference, this looks amazing, if you want I can make a video of the boot for anyone that is curious.

great work and idea on this @dmiart

TheSin-, 2020-08-19:
https://youtu.be/uOcAkeExMWU

dmiart, 2020-08-19:
Yeah, this is why I had the grey rectangle that was basically a copy paste of the original but with different colors to fit all resolutions. But it does look much better with all grey tbh.

UnchartedBull, 2020-08-19:
Holy sh*t that looks awesome! Kind of looks like a fully integrated system now, rather than something hacky. Really appreciate your efforts! I'll try to integrate that into the installation script and merge it over in the wiki so other people can also adjust there systems :)

TheSin-, 2020-08-19:
yeah it looks really good, it was a great idea @dmiart had.