UnchartedBull/OctoDash #910: Ubuntu Support

JNSmith1717, 2020-08-16:
Anyway that this would work with Ubuntu Server 18.04 LTS?  I run Ocotprint off a old laptop board with Ubuntu Server 18.04 LTS and would like to have a 5in hdmi display connected directly to it.  Only other way is to use a Raspberrry pi zero to just have OctoDash on it  but i see that don't work either.

dmiart, 2020-08-17:
I suspect you can just compile it on x86. 

https://github.com/UnchartedBull/OctoDash/blob/master/CONTRIBUTING.md

UnchartedBull, 2020-08-18:
There is already an amd64 version attached to the releases, which should work with x86. You just need to adjust the installer script to point to the other archive / install manually. Let me know if this works.

UnchartedBull, 2020-08-18:
Wrong button, sorry

JNSmith1717, 2020-08-20:
Amd64 would work for me actullay how do I getbthis

JNSmith1717, 2020-08-20:
Sorry kinda of new to using github

UnchartedBull, 2020-08-20:
It's right at the bottom of the releases: https://github.com/UnchartedBull/OctoDash/releases/tag/v2.0.0 you can either click & download or execute `wget https://github.com/UnchartedBull/OctoDash/releases/download/v2.0.0/octodash_2.0.0_amd64.deb`, which will download the file. Can be installed with `sudo dpkg -i octodash_2.0.0_amd64.deb`

closed 2020-09-15 by UnchartedBull