UnchartedBull/OctoDash #914: French translation

titis14, 2020-08-17:
Hello I installed octodash on my pi with a 7 inch screen in 1024x800 I am delighted with this application. I would like to know if the French translation is possible? I would like to contribute to translate the application. Is this easily done from a language file or is everything hard in the code?
Thank you

UnchartedBull, 2020-08-18:
Currently everything is hardcoded unfortunately. But having an i18n translation file is definitely on the list of improvements now. I'm changing the issues title and would come back to you for a french translation once this is done. Would this work for you?

titis14, 2020-08-19:
Hello,
Ok great I'm waiting for your return to be able to do the translation :)

Der-WU, 2020-09-01:
Hi there.
I can translate it to German if it rdy. :)

Protoncek, 2020-09-08:
I can translate to Slovenian...

Zero-Grav, 2020-10-23:
Hi 
Could you please give me a language translation tutorial？！

TroisSix, 2021-01-18:
Also interested to help traduct in french :)

pciavald, 2021-03-21:
I have located and translated all hard-coded strings of the app to french, i can implement this feature if the correct structure is implemented, or at least advised (long time since i worked on an angular app, i would not be the right person to choose the plugins and file architecture).

pciavald, 2021-03-22:
If no indication is given, i will start tomorrow. We have just rebased our branch to your latest main, and most changes are in conflict so i'll be faster implementing it correctly than fixing the merge :)

UnchartedBull, 2021-03-23:
Didn't had a thorough look at the possibilities yet, but `ngx-translate` seems promising, but I'm open to other libraries / the default i18n localisation as well. It should be able to:

- Handle untranslated content (i.e. adding a new feature with no translation available yet)
- Should work in serve mode (at least in english)
- Don't dramatically increase build times

If you find something let me know and I can have a look at that :)

pciavald, 2021-03-23:
I'm following this at the moment: https://angular.io/guide/i18n

pciavald, 2021-03-23:
Mostly done, will PR tomorrow

UnchartedBull, 2021-04-11:
Thanks to @pciavald this has been implemented and a french translation is also available. If anyone of you want to add another translation please feel free to do so. A how-to is available in the CONTRIBUTING file. Thanks :)

closed 2021-04-11 by UnchartedBull

Max3155, 2021-08-02:
> Hi there.
> I can translate it to German if it rdy. :)

Hi, can you help me with the translation to germany? I need this for a projekt in a few days.

Max3155, 2021-08-02:
Or can someone else help me who is familiar with it?

jneilliii, 2021-08-02:
Are you having issues with these instructions? https://github.com/UnchartedBull/OctoDash/blob/main/CONTRIBUTING.md#add-a-new-language

Max3155, 2021-08-02:
Yes I can't follow the instructions properly somehow. Is there a video or something?

jneilliii, 2021-08-02:
none that I'm aware of. Have you at least got to the point where you have a german xlf file?

Max3155, 2021-08-02:
No, unfortunately not. The npm somehow doesn't work either. No matter what command I enter nothing comes back

jneilliii, 2021-08-02:
you can just copy this file and fill in the "target" sections for the german equivalent. https://raw.githubusercontent.com/UnchartedBull/OctoDash/main/src/locale/messages.fr.xlf

jneilliii, 2021-08-02:
once you have that I might be able to build you a version tonight that will work.

Max3155, 2021-08-02:
Okay I understand that. How do I go on then?

Max3155, 2021-08-02:
okay thank you very much

Max3155, 2021-08-02:
Can you then show me how it works?

jneilliii, 2021-08-02:
I've never done it, so would have to figure that part out, but have a local development setup of OctoDash at home. Pretty sure it's just a bunch of copying/pasting of source files in OctoDash to add the language.

Max3155, 2021-08-02:
okay, I'll translate it now

jneilliii, 2021-08-02:
Just a heads-up, I won't be home for about 6 hours, but can definitely try the process for you then. I'll keep screenshots and such to document the process for others.

Max3155, 2021-08-02:
Okay, Thank you

Max3155, 2021-08-02:
Okay, I'm ready with the file.

UnchartedBull, 2021-08-02:
Please share the file here (just drag&drop) or create a PR

Max3155, 2021-08-02:
Okay it's not perfect yet, but i will optimize it when it works

[locale_DE.zip](https://github.com/UnchartedBull/OctoDash/files/6918520/locale_DE.zip)

Max3155, 2021-08-02:
Could you send me a short video recording in which you translate it with the file?

jneilliii, 2021-08-02:
I think the line needs to be updated.

```
<file original="ng2.template" datatype="plaintext" source-language="en" target-language="fr">
```

to 

```
<file original="ng2.template" datatype="plaintext" source-language="en" target-language="de">
```

UnchartedBull, 2021-08-02:
https://github.com/UnchartedBull/OctoDash/tree/feature/german-translation I created this branch. Everything is working there. You just need to execute `ng serve --configuration=de & npm run electron:serve:big` to get the development server up and running. I'll attach a deb file once the build is through.

On a site note: The translation do need a bit of work, not a huge fan of "Sie" and I think "Du" might be better suited for something like OctoDash, plus some texts are fairly long, so they might not fit in their designated spots.

[image]

UnchartedBull, 2021-08-02:
@jneilliii updated that in the linked branch already.

UnchartedBull, 2021-08-02:
https://drive.google.com/file/d/1ZEMsNEkgBzbXCeqP38lXdsOoH-Gs44Ys/view?usp=sharing here is the deb with the included german translation

Max3155, 2021-08-02:
Okay, thank you very much, I'll test it out and then improve the translations again.

Max3155, 2021-08-02:
what do I have to do now?

[image]

Max3155, 2021-08-02:
How can I actually create such a .deb? and then how do i use it?

jneilliii, 2021-08-02:
I have github actions on my fork of OctoDash that will build the deb files. It's part of the packaging process. Once you have the deb file, you transfer it to your pi and use the command `sudo dpkg -i ./octodash.deb` from the folder you saved to via SSH or the terminal.

jneilliii, 2021-08-02:
`npm run pack` will package into a deb file. not sure if you have to pass additional parameters for a different language or not to that command.

Max3155, 2021-08-02:
Okay and how i can start octodash now?

Max3155, 2021-08-02:
[image]

jneilliii, 2021-08-02:
did you run through the standard install script first? I would do that and then install over the version it installs with the de one. then run `sudo service getty@tty1 restart` to have it restart.

Max3155, 2021-08-03:
Yes now it worked. Many Thanks. Can I actually personalize the symbols?

Max3155, 2021-08-03:
Can one of you show me how I can create the .deb?

jneilliii, 2021-08-03:
Like I mentioned before, `npm run pack` from the folder where the code is will compile the code into a deb file. I do not know if you have to pass any parameters to that for specific language or not.

jneilliii, 2021-08-03:
https://github.com/UnchartedBull/OctoDash/blob/main/CONTRIBUTING.md#setup

Max3155, 2021-08-03:
npm run pack always creates a .tgz file for me

jneilliii, 2021-08-03:
that's a compressed file. should be able to be extracted. on windows I use 7 zip.

Max3155, 2021-08-03:
Yes I know, but I need a .deb file

Max3155, 2021-08-03:
> https://drive.google.com/file/d/1ZEMsNEkgBzbXCeqP38lXdsOoH-Gs44Ys/view?usp=sharing here is the deb with the included german translation

I will create this deb file

jneilliii, 2021-08-03:
once you extract the tgz is there no deb file inside?

Max3155, 2021-08-03:
yes unfortunately no deb file
[image]

jneilliii, 2021-08-03:
what's in the build folder?

Max3155, 2021-08-03:
icon.png

Max3155, 2021-08-04:
Do you know the control file for the deb creator?

UnchartedBull, 2021-08-04:
The `package.json` file controls the deb creation. It should create a deb archive when running `npm run pack` though. Did the command return any errors? It's also not advised to run the deb file creation on the Pi itself, because it's not powerful enough and can run out of memory during the build proces.

Also please either open a new issue or continue this on the discord. We're pinging other people here for a completely unrelated issue. Thanks.

UnchartedBull, 2021-08-04:
Discord dev or support channel probably is your best bet.

PR #1610 (2021-04-11): i18n + FR
- add package @angular/localize
- add command `npm run locale:extract`
- add command to update translation files  `npm run locale:update`
- add setting to generate locales in serve and build modes
- add i18n and $localize markers to all locale strings with custom IDs
- add `fr` translation
- use host's language by default (env `LANG`)
- update readme.md and contriburing.md

### TODO

- ~~remove yarn.lock from PR~~
- ~~correct fr translation~~
- ~~make sure it builds correctly~~
- ~~add localization documentation~~
- ~~replace added spans by ng-containers~~
- ~~set customIDs~~

Fixes  [... cut]