UnchartedBull/OctoDash #916: Predefined Actions does not work [!WEB]

Dennis206, 2020-08-18:
I tried to use one of the predefined actions on the Octodash.

I'm running the octodash on the same Pi as I'm running octoprint from. Attached is a 7" touch display.
On the octodash GUI I set the action under -> Moinscreen->Setting->octodash

I typed in the predefined action similar to the kill button --> [!WEB]http://google.de:80

When I try to connect to google, it only opens the iframe and rides loading website... but nothing happens.

what I tried: 
- several combinations of URL with and without ports
- "[!WEB]http://google.de"
- "[!WEB]<http://google.de>"
- [!WEB]http://google.de
- [!WEB]<http://google.de>"
- [!WEB]google.de
- [!WEB]google.de:80

What I would like to do is to open the Octoprint website so use the "cancel object" plugin.
I know that i have to put the localhost or the IP of the Octoprint server in that action.
While nothing worked I tried first to connect to any other website.

I checked that the Pi itself is connected to the www twice.

dmiart, 2020-08-18:
Can confirm. Wanted to have an octoprint page accessible from there, but doesn't seem to work. What's interesting though I had it working when I killed the octoprint service. It was showing the instructions page on how to check that on octoprint service. I suspect the !WEB doesn't like the javascript. I haven't tested pages without javascript yet though.

UnchartedBull, 2020-08-19:
google won't work, since google is actively blocking iFrames. The web view is nothing more than a fullscreen iFrame, which works fine for camera streams / displaying simple data and other stuff. If you want the full blown OctoPrint to work within the iFrame this may be a starting point: https://community.octoprint.org/t/multiple-octoprint-instances-in-one-webpage-html-using-iframe-doesnt-work/18872 (there is a screenshot third post). Unfortunately there is nothing I can do about that, as the website you're trying to open does decide whether or not it works in an iFrame.

closed 2020-08-19 by UnchartedBull