UnchartedBull/OctoDash #918: Raspi cam liveview

iMike78, 2020-08-18:
**Is your feature request related to a problem? Please describe.**
It's not a problem, just a feature request...

**Describe the solution you'd like**
That would be a great function if we can check the cam picture or the live video through OctoDash. Some of us keep the printer under tent, or in a cabinet, and sometimes needs to be checked what's happening inside.

**Additional context**
My idea is for example the live video option with a "subtitle" which is contains the status of the printer (temps and job percentage)

UnchartedBull, 2020-08-19:
This will be tracked in #828. That issue will enable you to open the control view during print, on which you could assign your camera to a custom action. This wouldn't include the subtitle you're talking about. This is quite an interesting idea on its own, so I'm hijacking your ticket here and will solely focus on the subtitle, while opening the camera during print will be tracked in the other issue. Hopes that makes sense :)

iMike78, 2020-08-20:
Thank you, the custom command is working fine for now. I'm so excited to see this option in V3 😉
[image]
[image]

RedSquirrell, 2020-10-06:
This looks great allready good job. But is there any way to scale the iframe/screen so it would fit on the screen?

UnchartedBull, 2020-10-09:
It should already be fullscreen, isn't it? Or is yours too big to fit on the screen?

RedSquirrell, 2020-10-09:
Yes mine is too big to fit on the screen.

UnchartedBull, 2020-10-15:
Hmm that certainly is weird. What is your camera resolution compared to your screen resolution?

Yordan1976, 2020-10-18:
I use a Hyperpixel 4 with resolution 800x480 and a stream with 1280x960 and have to use the scrollbars. Unscaled is good to set the camera focus but scaled down would be better for checking the orientation.

3dattic, 2020-10-21:
Hi, I would like to test this on the Adafruit 3.5" TFT. Where is the Json file is located, I could not find it?

I am a noob here so step by step will be help ful and for others until this is part of the revised build.
Also where did the ICON come from?

3dattic, 2020-10-21:
Looking at the code i persume i need to do the following

"customActions": 
[{
    "icon": fas fa-video,
    "command": http://192.168.1.120/#control-fullscreen-open,
    "color": #0000FF (blue color)
},
...

3dattic, 2020-10-21:
what am i doing wrong here
chmod 777 config.json
nano config.json
# comment out the vertical ruler and add above with the additional bits 

{
"icon": "fas fa-video",
"command": "http://192.168.1.120/#control-fullscreen-open",
"color": "#0000FF",
"confirm": false
"exit": true,
},
save file in nano 
chmod file config.json 444 and then reboot, but when OctoDash starts again it looks like a fresh install, i have to add the API key or allow it in Octoprint but the config.json file reverts to the original file?

I'm trying to understand what i did wrong here?

UnchartedBull, 2020-10-21:
The command won't work like that. needs to be:

```
{
"icon": "fas fa-video",
"command": "[!WEB]http://192.168.1.120/#control-fullscreen-open",
"color": "#0000FF",
"confirm": false,
"exit": true
}
```

I wouldn't chmod/chown the file, that most certainly messed OctoDash up. The plugin loading the config is setting the correct permissions while creating, once you tinker with them it won't read the config properly and you'll see the setup assistant again.

3dattic, 2020-10-21:
How would i add the above? when i try and use nano to alter the file it says I don't have permissions when I try and save it?

thank you for the correct code, i missed out the [!Web]

iMike78, 2020-10-21:
> Thank you for the correct code, i missed out the [!Web],
> 
> How would I add the above? when I try and use nano to alter the file it says I don't have permissions when I try and save it won't close, so i have to disgard the chenges :-(
> 
> when I open the config.json file it says in read text this file is unwritable

You can change any of the custom commands from under the UI
Settings -> octodash
and scroll down

3dattic, 2020-10-21:
hi iMike78,
I just looked at your suggestion and see what to do, my question is do i need to add the [!WEB] in the gcode/Action box?
if you can add a screen shot of what it looks like on your OctoDash that would be helpful, thank you

iMike78, 2020-10-21:
> hi iMike78,
> I just looked at your suggestion and see what to do, my question is do i need to add the [!WEB] in the gcode/Action box?
> if you can add a screen shot of what it looks like on your OctoDash that would be helpful, thank you

I'm printing at the moment, so I can't make a screenshot as I can't open the settings  :)
You have to type in like this:
[!WEB]http://192.168.1.120/#control-fullscreen-open

3dattic, 2020-10-21:
thanks iMike78, i've made the changes but it does not work. When i click the video icon it justs jumps back to the main screen :-(

[image]
[image]

iMike78, 2020-10-21:
Tick out the "Return to home" ;└)

3dattic, 2020-10-21:
I'm getting closer and made the change by unchecking Retrun to home. \but now it sits on a "loading website" i think it will be looking to the logon info and not picking up the API key?

[image]

3dattic, 2020-10-21:
I saw that and changed it back after i posted to you, good observation!!

iMike78, 2020-10-21:
> 
> 
> I'm getting closer and made the change by unchecking Retrun to home. \but now it sits on a "loading website" i think it will be looking to the logon info and not picking up the API key?
> 
Or - which could be the real reason - your video feed URL is no good

3dattic, 2020-10-22:
I thought of that and changed it to 
http://octopi.local port 80, 8080 & port 5000 - all no good
I can navitgate on a browser from my computer to http://192.168.1.120/#control-fullscreen-open, when I do that the browser looks like it has to log into the Octopi first? that is why i was thinking the API key was not being picked up?

the other thing is to try te IPv6 address shown in the ssh terminal
http://[2a00:23c5:df84:<redacted>:192f:225d]

3dattic, 2020-10-22:
An oberservation on the config.json file when viewing in Terminal. The API key is not correct, when i updated to the latest version 2.1.0 and bootup, you are shown the start up screen and run through some configuration steps, I chose to have the API key approved from my laptop browser and did not manually type this into the Raspberrypi. This picked the application API key and not the current Global API key shown in Octoprint, so im thinking this is a possible reason, however I can control the printer still if i change the API in the config file to either one.

this time when i reboot i don't get the "Hey There" welcome screen?

UnchartedBull, 2020-10-22:
Ok maybe to clear a few things up:

- OctoDash doesn't require the global API Key, the application key that is requested during setup is more than sufficient
- OctoDash won't use your OctoPrint API key to authenticate on a different website, in fact OctoDash will never give out your API Key to anything as this can be a huge security risk (from v2.1.0 onwards it won't show it in the settings either)
- The URL you are using is pointing to the OctoPrint webinterface which isn't ideal since it is loading a lot of stuff that isn't required to just view a webcam stream.
- You need to allow embedding in OctoPrint to be able to view it from within OctoDash (https://community.octoprint.org/t/multiple-octoprint-instances-in-one-webpage-html-using-iframe-doesnt-work/18872/2)
- You probably want to use the integrated website from mjpg-streamer which doesn't require any authentication:  `http://192.168.1.120/webcam/?action=stream`

3dattic, 2020-10-22:
Hi, you beat me to the punch UnchartedBull, I tried the setting in the OctoPrint Webcam settings and it worked great

http://192.168.1.120/webcam/?action=stream

[image]

mikulik86, 2021-01-29:
@Yordan1976, @RedSquirrell did you somehow resolve the issue of scaling the iframe to fit on the screen? My stream is also bigger res than the LCD.
Thanks

amoose136, 2021-02-16:
It would be useful if this could somehow support selection of a camera from multiple cameras.  (https://github.com/mikedmor/OctoPrint_MultiCam) It's nice to have a nozzle camera for upclose inspection and a wider view camera for a timelapse. A nozzle camera is useful on the pi screen to check that the first layer is going down nicely but you might as well preview the wide camera view as well to make sure nothing is out of whack at the start of the print.

UnchartedBull, 2021-02-17:
would you like to have them running in parallel? That probably won't be included, but there will be the option to have 2 buttons on the custom actions where each opens a separate camera view.

Adjuro, 2021-05-22:
Hi all, 
I believe I am wondering the same. I have 2 cameras plugged and I would like to access them trough Octodash custom action.
But when I do so, the stream is way too big for my screen. I can scroll to see what I want but it is not handy at all.
Additionally, it is nearly impossible for me to close the window with the help of the red cross.

I wonder if there is a way to rescale the stream directly from the command line.
                                {
                                        "icon": "video",
                                        "command": "[!WEB]http://octoprint.local:8081/?action=stream",
                                        "color": "#8c7ae6",
                                        "confirm": false,
                                        "exit": false
                                },
                                {
                                        "icon": "video",
                                        "command": "[!WEB]http://octoprint.local:8080/?action=stream",
                                        "color": "#f5f6fa",
                                        "confirm": false,
                                        "exit": false
                                }

jneilliii, 2021-05-22:
My integrated webcam page in the OctoDash Companion plugin can resolve the scaling issue, and the goal is to add multi-cam support to allow you to switch between them with a drop-down list. To use, after installing the plugin use the Custom Actions to add `[!webcam]` from the drop-down and it will pre-fill in the full path to the correct URL, you may wish to edit the base URL, but should work as programmed on a octopi install with OctoDash running on the same pi. In order for the multiple cams to work, currently planning on using the multicam plugin settings for the different cams, assuming that's the most commonly used plugin for this setup. Let me know if you are using something else and I can account for that as well.

Adjuro, 2021-05-22:
Thank you @jneilliii.
I tried your way. I already had the companion plugin installed and running. I am also using the multicam plugin for my 2 cameras.

Using the [!webcam]  option does solve indeed the scaling issue. I can see the entire stream from the web cam and I do not need to scroll  anymore.
But I am still not able to close the window with the top cross. I am not sure if this is a bug or else.

Now I wonder how can I access to the second webcam ? This features is not yet available, do I understand correctly ?

The command [!WEB]http://localhost/plugin/octodashcompanion/webcam links to the default camera, in my case the USB camera. 

[image]
[image]
[image]

jneilliii, 2021-05-22:
It's a feature that's not yet available. Was going to work on it this weekend.

jneilliii, 2021-05-22:
I've got multicam support available in my current dev branch if you want to try it out and report back. You can install by copying/pasting the URL below in plugin manager > get more > ...from URL and clicking Install.

```
https://github.com/jneilliii/OctoPrint-OctoDashCompanion/archive/refs/heads/dev.zip
```

Adjuro, 2021-06-01:
Hi,

I will try to check it out then, I'll keep you posted.

jneilliii, 2021-06-01:
Just released the multicam support version 0.0.5.

Adjuro, 2021-06-01:
Ok got it and tested it

So far, using the companion webcam option I can access to my main camera (raspberry pi camera).

Still I get the video stream that is too big for my screen (configured resolution for the camera) so I still need  to scroll left and right up and down to see the center of my webcam. But it is working!

I need to test the 2nd camera but cannot do it right now.

[image]

jneilliii, 2021-06-01:
That's just the direct stream url, should be `[!WEB]http://octoshield.local:8080/plugin/octodashcompanion/webcam`

Adjuro, 2021-06-01:
So I tested my second camera and it is working with your url.
But to be honest, I don't see any difference if I use "plugin/octodashcompanion/webcam" or "?action=stream".
The video stream are in both cases too big for my screen.

But at least it work so good work.
Thanks a lot for the support.

[image]
[image]

jneilliii, 2021-06-01:
scrolling is unexpected for sure. do you have the muticam plugin installed? if so you should see links in the top of the page to switch between them, try adding `?webcam=1` and `?webcam=2` to these URLs based on where they exist in the multicam plugin list and see if those scrollbars are still there or not.

Adjuro, 2021-06-02:
Yes I do have multicam up and running.
[image]

Sorry but I am not sure to understand how i should implement the " ?webcam=1 " command .[!WEB]http://192.168.1.107:8080/plugin/octodashcompanion/webcam?webcam=1 ?

Anyway now with my setup i cannot access to the stream anymore.
[image]
[image]

I am stuck on a "loading website .. " page and on the top left corner a barely readable text saying : "400 : Not found ! No file extension found"

jneilliii, 2021-06-02:
Yeah, wrong address...those 8080 and 8081 are needed for the stream, but not for the plugin. use this instead...unless you are using a separate pi from OctoPrint for your cameras.

`[!WEB]http://192.168.0.107/plugin/octodashcompanion/webcam?webcam=1`

Adjuro, 2021-06-03:
And the magic happens ! 
It works now perfectly.
I can see the entire video strim and it is fitted to my screen size.
It works for both of my webcams !

Thank you very much for your support.

Really good work!

jneilliii, 2021-06-03:
glad we finally got it worked out for you.

Adjuro, 2021-06-05:
One remark, I noticed that when octodash is connected to the printer, the webcam is not accessible trough the custom action.
Is it normal (something like safety feature) ?
Or is it a bug ?

jneilliii, 2021-06-05:
That doesn't seem to be the case here for me. The webcam works no matter what the state. Do you mean while printing you can't get to the custom actions?

Adjuro, 2021-06-07:
Well if Octoprint/Octodash are on but not connected to the printer (sleep mode) I can access to the web cam.
If I connect to the printer, then as soon as I try to access them, the iframe is opening/closing immediatly.

It does not matter if I am printing or not.

jneilliii, 2021-06-07:
That is odd, I just double-checked and cannot reproduce this issue.