UnchartedBull/OctoDash #919: Change G_CODE for babysteping

Emented, 2020-08-19:
Hello guys! 
I've installed octodash andI really like it, but i use klipper and it has another comands for babystepping during a print(
So the question is, how ti change them, where are located files with this commands?

dmiart, 2020-08-19:
https://github.com/UnchartedBull/OctoDash/blob/330f9ae301918d2fd5a97f9f12f74a3361c1dec4/src/app/print-control/print-control.component.ts#L190

I believe it's this one.

You might also be able to make a MCODE 290 macro. Not sure whether macros can accept input parameters though.

Emented, 2020-08-19:
> https://github.com/UnchartedBull/OctoDash/blob/330f9ae301918d2fd5a97f9f12f74a3361c1dec4/src/app/print-control/print-control.component.ts#L190
> 
> 
> 
> I believe it's this one.
> 
> 
> 
> You might also be able to make a MCODE 290 macro. Not sure whether macros can accept input parameters though. 
> 
> 
> 
> 

So, how can I change this file on my PI?

dmiart, 2020-08-19:
I fairly new to this framework, but if I am not mistaken, all files are compressed inside /opt/OctoDash/resources/app.asar. However you can make your own debian package from source. 

https://github.com/UnchartedBull/OctoDash/blob/master/CONTRIBUTING.md

UnchartedBull, 2020-08-19:
Even if you'd manage to unpack it you probably won't find that line. Angular does a lot of magic before packaging everything (minifying, treeshaking and other optimizations), so this line won't look the same. Rebuilding everything is your best guess here. I can also add a config option to a later version, so users can define there baby-stepping commands, but that will take a while.

Emented, 2020-08-19:
> Even if you'd manage to unpack it you probably won't find that line. Angular does a lot of magic before packaging everything (minifying, treeshaking and other optimizations), so this line won't look [...]

I'm really new to all of this stuff, so I'l try to make a macro, and wait for the update, because I don't understand how to do it on my own)

whoim2, 2020-09-25:
Good idea - configurable g-codes and titles for babystep screen. I want use klipper command "ajust", no absolute value, and my values (0.05, 0.01)

UnchartedBull, 2020-09-28:
Custom values probably won't be added but the next release should at least include the option to replace `M290 Z` with something different.

whoim2, 2020-09-28:
> Custom values probably won't be added but the next release should at least include the option to replace `M290 Z` with something different.

Only one command? Relative variants not in plan?
For each button custom command

UnchartedBull, 2020-09-28:
I won't have the time to implement a custom command for each of those buttons. Feel free to open a PR though :)

whoim2, 2020-09-29:
> I won't have the time to implement a custom command for each of those buttons. Feel free to open a PR though :)

Teach me build your code on rpi or ubuntu?)

UnchartedBull, 2020-10-05:
https://github.com/UnchartedBull/OctoDash/blob/master/CONTRIBUTING.md

PR #1043 (2020-10-05): customizable babystep gcode
Closes #919.
closed 2020-10-05 by UnchartedBull