UnchartedBull/OctoDash #922: Get printer profile from OctoPrint instead of during config

TheSin-, 2020-08-19:
The printer profile is already exposed via the OctoPrint API, it would make sense to use and check it on interval incase the printerProfile changes.  I know personally I have 3 profiles, single nozzle, dual nozzle and Palette.  Be nice for OctoDash to know it's changes and update the display (once I finish the multi extruder setup).

This would also reduce the amount of input required during config to just the API URL and API Key.

I hope to start work on this one and add a PR for it in the near future, I'm still setting up my dev env for it all, and my angular is intermediate at best so it might take me some time.  But I think this will help get OctoDash users setup and working much quicker.

UnchartedBull, 2020-08-19:
Rudimentary support is already done #856, so the printer profile is known to OctoDash, it currently just ignores everything except for axis inversion. It would make sense to move this to a separate service though, since it would be used by multiple components.

Edit: Commented that on the wrong issue, sorry.

TheSin-, 2020-08-23:
Okay so looking at printerprofile.service and printer.service

I think printerprofile should be on a loop maybe not as often as printer but same idea.

It should not use _default it should get the list and look for current: true, this was an issue for me I switched profiles and things didn't match up now I see why.

This is also where we can see all the details and we will need to add more config options.

we can get and set the

name
axes speed
extruder count
heated bed
heated enclosure

I'd like to extend the config to include the last 3 and we can have it update on an interval so when you switch profiles the ui can update as well.  This will already remove a ton from the initial setup.  I'm still working on the bonjour list to select the api during the onboarding.

Not this is where you check if printer is Offline.  But this should be done in printer.service, and we can set an app global so that we can stop the cards and allow entry to the config while it's offline.  This will stop lots of grief for many of us.  I know I like to do lots of config and checks while things are offline.  Plus the filament storage could still be online and avail.

in printer services just look for "Printer is not operational"

api not avail should be a different state IMHO

UnchartedBull, 2020-08-24:
Why do we want to add this to the config though? If we read it from the API semi-regularly (like every odd minute or so) we can just keep those values in memory and expose them via the configuration service. This was they will look like normal config files, but won't be persisted. Or is there any other benefit I'm missing here?

The printer-turned-off state needs some major rework yes. OctoDash can already detect if the printer isn't connected or if there is an error with the API (it will only enter sleep mode if the printer is disconnected). What's left now is to unlock some functionality in that case. I personally would push that behind the OctoPrint push update, since that requires a major rewrite on it's own, so adding the sleep improvements can be done along that.

TheSin-, 2020-08-24:
By Config I meant the config object so it's accessible by all modules.  I don't mean the flat file, just the object, I fell all this stuff should get removed from the json file.

Unless you want to have an other object to store in?

I agree, moving to push will make everything else easier moving forward.

UnchartedBull, 2020-08-24:
Then we are on the same page there. I think everything that can be retrieved from the API should be stored in OctoPrint and shouldn't be stored in OctoDash (just temporarily inside the config object to make access easier).

TheSin-, 2020-08-31:
Did you want to use this to track the change over to push, I might look into that one once we are doing this the config changes.

TheSin-, 2020-10-02:
I'm going to start on this one and the installer to get that boot splash this weekend I think.  Do you think this is a good place to spend time or is there a better place?

UnchartedBull, 2020-10-02:
Both are pretty good issues :) In general you can take any issue that looks like fun to you, just comment something like "I'll take that" and off you go. Thanks for all your work!

UnchartedBull, 2021-01-04:
Should probably somehow indicate if printer is virtual as described in linked issue.