UnchartedBull/OctoDash #928: When changing screen orientation the touch points are not moving

silverbrewer07, 2020-08-21:
**What doesn't work?**
When changing screen orientation the touch points are not moving. In my case I am using a 2.8 inch screen which is natively set to be portrait at 240x320 upon rotate to landscape the ui looks good but the touch screen inputs are where the icons were originally. 

**What did you already try?**
Tried to update the software. I've tried rotating several different ways with the same results. 

**General Information:**
Octodash v2
**Additional context**
Add any other context or screenshots about the feature request here.

stibabs, 2020-08-23:
Same problem...

UnchartedBull, 2020-08-24:
I guess you need to rotate your touch input device as well (possibly recalibrate it). In some cases the touch interface doesn't get rotated if the image is rotated.

I don't have any influence on how the touch is handled, since that is all handled by Raspbian and electron.

AdmiralMichael, 2020-08-30:
Try this:

[](https://www.raspberrypi-spy.co.uk/2017/11/how-to-rotate-the-raspberry-pi-display-output/) 

But use lcd_rotate instead as display_rotate doesn’t rotate the touchscreen input.

dexter323i, 2020-09-05:
If you use a "generic" chinese display, and maybe this helps:
https://github.com/goodtft/LCD-show

UnchartedBull, 2020-09-07:
Since there is no new activity here from OP I'm closing this. lcd_rotate should work.

closed 2020-09-07 by UnchartedBull

chickenwoman, 2021-01-29:
If you're using the BTT pi tft5 you need to use lcd_rotate=2 and display_rotate=2 to flip it.

MrElie, 2022-03-02:
here is the solution for those who need it :

1st u need to go to /boot and edit config.txt
`sudo nano /boot/config.txt`

at the very bottom add:
`display_lcd_rotate=2`

Note:  The angle of rotation is counted clockwise. display_lcd_rotate=0 (landscape) display_lcd_rotate=1 （90 degress） display_lcd_rotate=2 （180 degrees） display_lcd_rotate=3 （270 degrees）

Note2: I'm using the 180 degrees option in this example so change to your liking.
save and
`sudo reboot now`

after it reboots you will see touchscreen not responsive because we just rotated the lcd display but not the touch screen. to rotate the touchscreen simply do the following steps:
go to:
`cd /usr/share/X11/xorg.conf.d`

find the libinput file by doing 
`ls`

u can now edit it will be something like:
`sudo nano 40-libinput.conf`

in the Section "InputClass" find the Identifier that says: "libinput touchscreencatchall" and below the line MatchIsTouchScreen "on" add the following line:
`Option "TransformationMatrix" "0 1 0 -1 0 1 0 0 1"`

Note1: 
```
90° = Option "TransformationMatrix" "0 1 0 -1 0 1 0 0 1"
180° = Option "TransformationMatrix" "-1 0 1 0 -1 1 0 0 1"
270° = Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"

```
Note2: Again i'm using the 180 degrees in this example change to your liking from Note1 section
Note3: The XY position of different touch screen's start point may be various, so the transformation matrix values of 90° and 270° may be the opposite. For models other than M505T, please try to adjust it yourself! This has been tested on the raspberry PI official 7 inch touchscreen.

save reboot everything should work properly 
`sudo reboot now`