UnchartedBull/OctoDash #940: Can't execute shutdown command (Predefined Action [!SHUTDOWN])

brtlzr, 2020-08-26:
The Predefined Action [!SHUTDOWN] does not work.
Following error message is shown:

**Can't execute shutdown command**
Http failure response from http:/:localhost:5000/api/system/commands/core/shutdown
403 Forbidden

**To Reproduce**
Edit the config.json file
I've replaced the KILL button with a power-off button with following code:
`				{
					"icon": "power-off",
					"command": "[!SHUTDOWN]",
					"color": "#e84118",
					"confirm": true,
					"exit": false
				}
`
Saved and rebooted my Pi.
Clicking the power button gives the 403-error.

**Expected behavior**
Clicking the power button sends the shutdown command and the Pi shuts down.

**General Information:**
 - Raspberry Pi 3 Model B Rev 1.2
 - Octopi image Version 0.17.0
 - OctoDash Version v2.0
 - OctoPrint Version 1.4.2

**Additional context**
My Pi is powered with a buck converter to my printer's power supply. I want to do a clean shutdown of the Pi before powering off my printer.

UnchartedBull, 2020-08-31:
Please check your OctoPrint ACL and whether you have setup the shutdown scripts correctly as outlined here: https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337 (Support restart/shutdown through OctoPrint's system menu)

dexter323i, 2020-09-05:
@brtlzr That dedicated user must be in the Admins group. Operator is not enough for rebooting.

UnchartedBull, 2020-09-07:
Probably what @dexter323i mentioned, please promote the OctoDash user to admin and try again.

closed 2020-09-15 by UnchartedBull

brtlzr, 2020-09-15:
I've just added the user to the admins group and that has fixed the issue.
Thanks for the support!