UnchartedBull/OctoDash #941: Can't retrieve Job Status while printing files that are in a folder on the SD card

brtlzr, 2020-08-26:
**Describe the bug**
I've got a "tests" folder on my SD card with some gcode files in it to do first layer tests etc.
When starting a print from the printer of one of the files in that folder OctoDash starts showing error messages.
it alternates between:

**Can't retrieve Job Status**
and
**Can't load thumbnail!**
Http failure response for http://localhost:5000/files/sdcard/baseline.gcode... 404 Not Found

Strange thing is that I am not using the thumbnail plugin.

**To Reproduce**
Create a folder on your SD card
Put a gcode file in the folder
Insert the SD card in your printer
Select print from SD on your printer and select the file in the subfolder
Look at the OctoDash screen

**Expected behavior**
OctoDash shows the job status

**General Information:**
- Raspberry Pi 3 Model B Rev 1.2
- Octopi image Version 0.17.0
- OctoDash Version v2.0
- OctoPrint Version 1.4.2
- Ender 5 pro with BL-touch and TH3D firmware

**Additional context**
I guess a bit of the path is missing. In my example it should ask for a job status of http://localhost:5000/files/sdcard/**tests**/baseline.gcode

UnchartedBull, 2020-08-31:
I'll have a look at that. Does the OctoPrint webinterface show the job status?

brtlzr, 2020-09-02:
I just tried it again.
The problem only occurs for files in a subfolder op the SD card.
Yes, Octoprint shows the job status.

UnchartedBull, 2020-09-07:
Ok, I'll have a look at that :)

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550
closed 2021-03-18 by UnchartedBull