UnchartedBull/OctoDash #945: Neopixel Support

meganssmith, 2020-08-26:
_**Is your feature request related to a problem? Please describe.**
NA

**Describe the solution you'd like**
It would be nice to have a button on octodash that would let us control the enclosure lights via the Neopixels control in the Enclosure Plugin or similar.    

**Link other projects**
https://github.com/vitormhenrique/OctoPrint-Enclosure

**Additional context**
NA

dmiart, 2020-08-26:
I replaced one of six action with a GCODE to turn on the lights as a temporary solution to this. 

In config replace one of them with 

                               {
                                        "icon": "lightbulb",
                                        "command": "M150 B200 R200 U200",
                                        "color": "#44bd32",
                                        "confirm": false,
                                        "exit": true
                                },

meganssmith, 2020-08-26:
I run the BTT SKR 1.3, so no real easy spot to put the neopixles on there.  I have seen where some move the BLtouch to Y-max and put the pixels on the servo pins, but with them working well on the Pi, I don't feel the need to sort out that mess in Marlin.

dmiart, 2020-08-26:
I think you can also re-use some of the monochrome LCD pins EXP to drive PWM on those strips. But yeah I agree it's a pain to set it up if there is no dedicated pinout on the board for it.

UnchartedBull, 2020-08-31:
Will be added as a custom action.

UnchartedBull, 2020-10-05:
Docs: https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions. Plese let me know if this works once the new release is ready. My LED strips are directly hooked up to the printers mainboard, so I can't test it.

meganssmith, 2020-10-05:
OK, I can do that, thank You!

closed 2020-10-05 by UnchartedBull

carcamerarig, 2020-10-20:
> I think you can also re-use some of the monochrome LCD pins EXP to drive PWM on those strips. Just need to find one pin and attach second pin to the ground since it's powered externally. But yeah I agree it's a pain to set it up if there is no dedicated pinout on the board for it. 

Would you be able to help me set this up on an Einsy board?

meganssmith, 2020-10-24:
OK, Test and it works! :)

PR #1044 (2020-10-05): add custom action for LED Strip
Closes #945.