UnchartedBull/OctoDash #955: Spool change freezes and crashed only when using a duet mainboard

acbaldwi, 2020-08-30:
One of my printers uses a duet mainboard, when i go into the filament change section octodash goes to a sleeping screen then after a minute bounces back to the main page, this does not happen when connected to a marlin based printer.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to filament
2. Click on skip
3. click skip (or wait for it to heat up)
4. Get to the "i'm done screen
5. crash

**Expected behavior**
Should allow me to push out filament

**Screenshots**
Working on marlin based
https://youtu.be/SP60EtFEEww

Broken on Duet based
https://youtu.be/7M91aeOVnvg

**General Information:**
 - Hardware Duet Wifi 2, connec ted to a raspberry pi which has octodash and a touchscreen on it
 - OS Info 
 - OctoDash Version [2.0)
 - OctoPrint Version [e.g. v1.4.2]

**Additional context**
I have 3 printers running octodash, the 2 marlin based ones work just fin with this feature

UnchartedBull, 2020-08-31:
Does that also happen if you let the nozzle heatup? I don't know how the temperature protection works on Duet boards. Please also turn off the Filament Plugin Manager plugin and try again.

acbaldwi, 2020-08-31:
It does it even if you let it preheat, i did finnally capture a error in the duet logs 
"8/31/2020, 10:02:11 AM | Error: Attempting to extrude with no tool selected.
-- | --"

acbaldwi, 2020-08-31:
I should also mention i updated duet to it's latest version 3.1 and it does the same as the older versions

UnchartedBull, 2020-09-02:
Do you use multiple extruders by any chance? https://docs.octoprint.org/en/master/api/printer.html#issue-a-tool-command this is the API I'm currently using to extrude filament and it sasys:

> Extrudes the given amount of filament from the currently selected tool. Additional parameters:

The weird thing is, why is there no tool selected?

acbaldwi, 2020-09-06:
You sir are a genius,

While i only run a single tool apparently octoprint assumes tool 0, and no others unless defined... where as duet (reprap) assumes nothing thus the issue. SHoudl anyone else run accross this simply add the below to your config.g
"T0                                      ; select first tool"

closed 2020-09-06 by acbaldwi