UnchartedBull/OctoDash #956: Uninstall instructions - no desktop environement

msteele999, 2020-08-30:
I recently installed OctoDash on my Intel Compute Stick running OctoPrint on Ubuntu 18.04 server - needless to say, I did not install a desktop environment and got it in my head that this was a web based Dashboard.

I'd like to cleanly remove everything from the install but I'm not sure where to start. 

Is there a clean linux command that will remove the plugin? 'apt list --installed' does not list octodash.

Thank you.

UnchartedBull, 2020-08-31:
`wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/remove.sh | bash` should remove everything. Maybe delete the `~/.config/octodash` folder as well.

closed 2020-08-31 by UnchartedBull

ArrowRaider, 2021-08-19:
This did not remove it from automatically starting.

ArrowRaider, 2021-08-19:
https://github.com/UnchartedBull/OctoDash/raw/master/scripts/remove.sh does not exist