UnchartedBull/OctoDash #957: UI state to reflect auto pause state from print.

hamid-elaosta, 2020-08-30:
**Describe the bug**
Pause/resume status isn't synchronised with printer status.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print which pauses itself (I use manual bed leveling wizard).
2. Print pauses itself.
3. Tap screen to show pause button
4. Tap Pause
5. Tap Resume

**Expected behavior**
Currently, resuming an auto-paused print is 3 taps, Ideally OctoDash would detect that the print paused and display a resume button for one button resume. Worst case should be two clicks, one to show the buttons, second to resume.

**Screenshots**
N/A

**General Information:**
- Raspberry Pi
 - OctoDash Version: 2.0.0
 - OctoPrint Version: 1.4.2

**Additional context**
I've made this a bug report because I believe the status should reflect the printer status (2 clicks to resume) at the very least, but ideally 1 click, but I think that would be a feature request to auto-display the paused state.

UnchartedBull, 2020-08-31:
I thought there was an issue for that already but I'm unable to find it. Will be included in the next release.

hamid-elaosta, 2020-08-31:
Thanks @UnchartedBull. I did take a look before raising it but didn't find one either.

PR #958 (2020-08-31): Show pause screen if not paused from OctoDash
Closes #957.
closed 2020-08-31 by UnchartedBull