UnchartedBull/OctoDash #962: Screen Saver

OCD3DP, 2020-09-01:
Maybe I've missed it somewhere, but is there a way to have the screen sleep or add a screen saver while printing or in standby? I forgot to turn my printer off last night, and now the control screen is mildly burned into my lcd screen.

Thanks in advance!

TheSin-, 2020-09-01:
I believe the wiki has info about enabling DPMS

https://github.com/UnchartedBull/OctoDash/wiki/Installation#start-octodash-automatic-on-boot

TheSin-, 2020-09-01:
Might be nice to have a control for this though, set it to off or a timeout incase you want to watch a print.  with set it should be controllable.

OCD3DP, 2020-09-01:
Thanks... yep, I'm not extremely software savy, so a lot of this stuff is like reading Sanskrit to me.

TheSin-, 2020-09-01:
the important parts are the set commands under

Modify DPMS and screensaver settings with a command

UnchartedBull, 2020-09-02:
Will be tracked in #410. Once that is done OctoDash can do the screensaver stuff with nice UI configuration. Until then please rely on dpms (change that in `~/.xinitrc`).

Btw are you using an OLED screen? Burn-in shouldn't happen on LCD screens, at least that's what I always though.

closed 2020-09-02 by UnchartedBull

OCD3DP, 2020-09-04:
@UnchartedBull I'm using a Waveshare 7inch HDMI LCD (C) Capacitive Touch Screen... I just Google searched, and found that someone else had burned screen issues, so I may just replace it with something different. I'll have to take a crash course in making changes to the `~/.xinitrc`

Thanks!

dsbaars, 2020-09-06:
I have the same screen and #410 will not solve this for us if it is only reducing the backlight. 

To manually turn it off using the console:
````bash
export DISPLAY=:0
xset dpms force off
````

UnchartedBull, 2020-09-07:
@dsbaars that is what is currently used for turning off the screen while sleeping. Why wouldn't that command work? I though all it does is set the screen brightness to 0 (which means turning off the backlight). Have you tried `echo 0 | sudo tee /sys/class/backlight/rpi_backlight/brightness`?

dsbaars, 2020-09-07:
It's an HDMI screen, which you can't control because `/sys/class/backlight` does not exist. I expect that only exist for screens connected through the DSI connector.

UnchartedBull, 2020-09-07:
Does `xbacklight -set 0` work with an HDMI screen?

dsbaars, 2020-09-07:
Nope:
````
pi@octopi:~ $ xbacklight -set 0
No outputs have backlight property
````

I also tried `xrandr --output HDMI-1 --brightness 0.0` which does not give any errors in the console but does not have any effect.

Maybe this output is also useful:
````
pi@octopi:~ $ xrandr --verbose
Screen 0: minimum 320 x 200, current 1024 x 600, maximum 7680 x 7680
HDMI-1 connected primary 1024x600+0+0 (0x42) normal (normal left inverted right x axis y axis) 0mm x 0mm
        Identifier: 0x40
        Timestamp:  64425925
        Subpixel:   unknown
        Gamma:      1.0:1.0:1.0
        Brightness: 1.0
        Clones:
        CRTC:       0
        CRTCs:      0
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter:
        Broadcast RGB: Automatic
                supported: Automatic, Full, Limited 16:235
        bottom margin: 0
                range: (0, 100)
        top margin: 0
                range: (0, 100)
        right margin: 0
                range: (0, 100)
        left margin: 0
                range: (0, 100)
        link-status: Good
                supported: Good, Bad
        CONNECTOR_ID: 54
                supported: 54
        non-desktop: 0
                range: (0, 1)
  FIXED_MODE (0x42) 49.000MHz +HSync +VSync *current +preferred
        h: width  1024 start 1064 end 1168 total 1312 skew    0 clock  37.35KHz
        v: height  600 start  603 end  610 total  624           clock  59.85Hz
````