UnchartedBull/OctoDash #963: Octodash not working after updating DisplayLayerProgress to 1.23.2

pilotw7, 2020-09-01:
**What doesn't work?**
I saw there was a new update to DLP and updated it, but after I updated the plugin Octodash now reads "Error (0) in the bottom right corner and says cannot retrieve status and cannot retrieve jobs. Everything was working fine prior to updating DLP. Is the current version of OctoDash not compatible with it?

**What did you already try?**
The problem persists through reboots and not sure what else to do. 

**General Information:**
 - Hardware Raspberry Pi 3 connected to a Monoprice Maker Ultimate
 - OS Info OctoPi Version 0.17.0
 - OctoDash Version 2.0.0
 - OctoPrint Version 1.4.2

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2020-09-02:
Should be working with the latest DLP (although I haven't tried it myself). Could you please double check the printers IP Address / URL? error(0) usually means that there is no response at all from OctoPrint. Does the webinterface still work?

pilotw7, 2020-09-02:
The web interface works fine and everything, its just Octodash that has stopped working.

thegarbz, 2020-09-05:
Just a suggestion can you reset the Display Layer Progress plugin settings? I've not seen your issue and am running all the same versions you are. The update to DLP (yesterday I did it) was uneventful.

pilotw7, 2020-09-06:
I tried doing a reset of the plugin settings, still getting the errors for "DisplayLayerProgress Unknown Error" and "Can't retrieve Jobs"

UnchartedBull, 2020-09-07:
That certainly is really weird. Could you please execute `curl -v http://localhost:5000/api/version` (replace localhost with your Raspberry Pis IP, if you execute it from another machine)? It should return something like this:

```
*   Trying xxx...
* TCP_NODELAY set
* Connected to xxx port 5000 (#0)
> GET /api/version HTTP/1.1
> Host: xxx:5000
> User-Agent: curl/7.64.1
> Accept: */*
> 
< HTTP/1.1 403 FORBIDDEN
< Content-Length: 234
< X-Robots-Tag: noindex, nofollow, noimageindex
< X-Content-Type-Options: nosniff
< Cache-Control: max-age=0
< X-Clacks-Overhead: GNU Terry Pratchett
< Content-Type: text/html; charset=utf-8
< X-Frame-Options: sameorigin
< 
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>403 Forbidden</title>
<h1>Forbidden</h1>
<p>You don't have the permission to access the requested resource. It is either read-protected or not readable by the server.</p>
* Connection #0 to host xxx left intact
* Closing connection 0

pilotw7, 2020-09-07:
When I tried doing the localhost command I got back:

*   Trying ::1...
* TCP_NODELAY set
* Expire in 149999 ms for 3 (transfer 0xaa0880)
* Expire in 200 ms for 4 (transfer 0xaa0880)
* connect to ::1 port 5000 failed: Connection refused
*   Trying 127.0.0.1...
* TCP_NODELAY set
* Expire in 149999 ms for 3 (transfer 0xaa0880)
* Connected to localhost (127.0.0.1) port 5000 (#0)
> GET /api/version HTTP/1.1
> Host: localhost:5000
> User-Agent: curl/7.64.0
> Accept: */*
>
< HTTP/1.1 403 FORBIDDEN
< Content-Length: 234
< X-Robots-Tag: noindex, nofollow, noimageindex
< X-Content-Type-Options: nosniff
< Cache-Control: max-age=0
< X-Clacks-Overhead: GNU Terry Pratchett
< Content-Type: text/html; charset=utf-8
< X-Frame-Options: sameorigin
<
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>403 Forbidden</title>
<h1>Forbidden</h1>
<p>You don't have the permission to access the requested resource. It is either read-protected or not readable by the server.</p>
* Connection #0 to host localhost left intact

pilotw7, 2020-09-07:
I'm not sure if this is related, but I'm also having issue with some other plugins, TP-Link Smartplug and Shutdown Printer, where usually they would send command to shutdown the Printer & TP-Link after print is done, but now it gets command issue where it times out and the Raspberry Pi only disconnects and errors.

pilotw7, 2020-09-08:
Is there any command to reset / uninstall Octodash?

UnchartedBull, 2020-09-11:
`wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/remove.sh | bash` should remove OctoDash. The curl command is showing that your API is working and at least somehow responding (the forbidden is fine, since you haven't supplied a token). That other plugins are having issues as well is indicating that something funny is going on with your OctoPrint installation.

pilotw7, 2020-09-12:
Okay, after uninstalling and reinstalling Octodash everything is working fine again! Thank you for the help!

closed 2020-09-14 by UnchartedBull