UnchartedBull/OctoDash #971: Text layout issue on printing screen

greendog99, 2020-09-03:
**Describe the bug**
Status/progress text and icons are in wrong position on screen, resulting in overlapping and clipped text.

**Screenshots**
[image]

**General Information:**
- RPi 4 w/ OctoPi
- 800x480 touchscreen
- OctoDash v2.0.0 special version to fix touch/drag issue #838 

**Additional context**
Screen display was working properly in the past. May be related to #838 fix?

greendog99, 2020-09-04:
Note: I've noticed that this only happens on some prints (so it may be unrelated to #838 after all). I'll try to narrow down the conditions where I see this happen...

greendog99, 2020-09-06:
I've done some testing and this appears to be a text wrapping issue due to the font size. In the above photo, if I manually change the printer's temperature to 199 (instead of 200), the layout corrects itself, since "199ºC/199ºC" takes up less horizontal space than "200ºC/200ºC". Hope this helps!

UnchartedBull, 2020-09-07:
That certainly doesn't look right. I'll have a look at that and check, whether I can reproduce it on my machine

PR #1041 (2020-10-05): Fix text wrap and switch to bonjour
Fixes #971.
closed 2020-10-05 by UnchartedBull