UnchartedBull/OctoDash #975: Issue with installation

talunceford, 2020-09-04:
Hello,

I am having a few issues installing octodash.  I am receiving a dpkg: dependency problems prevent configuration of octodash:armhf:
 octodash:armhf depends on libgtk-3-0.
 octodash:armhf depends on libnotify4.
 octodash:armhf depends on libnss3.
 octodash:armhf depends on libxss1.
 octodash:armhf depends on libxtst6.
 octodash:armhf depends on libatspi2.0-0.
 octodash:armhf depends on libappindicator3-1.
 octodash:armhf depends on libsecret-1-0.

dpkg: error processing package octodash:armhf (--install):
 dependency problems - leaving unconfigured
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for gnome-menus (3.31.4-3) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...

I've tried installing it manually and automatically via the install script and I get the same message.  I've tried fixing the dependencies and it ultimately uninstalls octodash:armhf.

Any help is very much appreciated.

Thank you for your time and assistance.

Tim

TheSin-, 2020-09-04:
type apt-get install -f, see what it says.

closed 2020-09-15 by UnchartedBull

talunceford, 2020-09-15:
Just to let you know, I got it to work.  The version of octoprint that came on my pi was too new.  I reverted to the 1.17.x version and it works like a champ!  Thank you!