UnchartedBull/OctoDash #979: Octodash Not Showing

Toxicl16, 2020-09-08:
Hi all. I just installed octodash and the program never shows. Here is some info. Any help would be greatly appreciated!
 
Octodash fails to load each time the Pi boots up. Manual start-up of octodash also fails. When starting ratpoison it says it cannot open display. The octodash command returns Segmentation Fault upon running it.

I have updated both octoprint and octodash to the latest versions. I have also installed the display drivers, and all of the boot up process is displayed on the screen. I have also tried to do the export DISPLAY=:0 and that also didn't work.

**General Information:**
 - Raspberry Pi 3 B
 - 3.5 inch TFT via GPIO. Resolution is 480x360 I believe
 - Latest update of Octoprint
 - Latest update of Octodash

Thanks for helping,
   Toxicl16

UnchartedBull, 2020-09-11:
Have you checked the troubleshooting guide: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#cannot-open-display-xinit-failing-

Toxicl16, 2020-09-11:
Hi, I just checked that wiki page. Sadly that didn't help me. When I run `ratpoison` I get the following issue: `Can't Open Display` The return of `octodash` is `Segmentation Fault`. So I attempted to start X via `xinit` and `startx` however I get the following Issue:
`X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
Current Operating System: Linux octopi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo quiet splash plymouth.ignore-serial-consoles
Build Date: 05 June 2019  12:49:54PM
xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri Sep 11 09:03:40 2020
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/4.19.75-v7+`

UnchartedBull, 2020-09-14:
according to this thread https://www.raspberrypi.org/forums/viewtopic.php?t=67334 this FATAL error isn't an issue. Is this the whole log, or is there more output?

Toxicl16, 2020-09-15:
So, I did some research on the other issues, and I found that some people had better success following a certain installation procedure. I followed it, and Octodash is working correctly now. Thanks for the help, and the great project!

Toxicl16

closed 2020-09-15 by Toxicl16

alexanderphoenix, 2020-09-26:
@Toxicl16 I seem to be having the same issue on a fresh install (of octoprint and octodash) on a RPi 3B.
Could you share what this "certain installation procedure" is, please?
Many thanks.

Toxicl16, 2020-09-27:
@alexanderphoenix I followed a comment in the following issue. Hope it helps!

https://github.com/UnchartedBull/OctoDash/issues/574