UnchartedBull/OctoDash #991: Octodash stuck on starting even though print is running

dani190, 2020-09-13:
**What doesn't work?**
Octodash is stuck on starting even though the print is running. Screen shows the correct temp and fan speeds and is responsive (I can change things like the fan speed while it's printing)

**What did you already try?**
Rebooting and ensuring I am on the latest version. The issue happens every few prints so it's not 100% consistent.

**General Information:**
 - Raspberry PI 3B+ with a 7" touchscreen
 - Raspbian buster - 4.19.75-v7+
 - OctoDash Version - Whatever the latest is. Ran upgrade script yesterday.
 - OctoPrint Version - 1.4.2

UnchartedBull, 2020-09-14:
Could you please check what the bottom right text says next time it occurs? Could be that your installation is using a different word for "printing" which is used to detect a print started from another ui. Does this only occur if you start a print from the browser, or also if you start it from octodash?

dani190, 2020-09-14:
It just says starting and stays there.

What’s odd is if I cancel the print and start it again with the same file it will work. All of this is done via the octoprint webpage. I have not tried starting a print from octodash yet.

antonyburden, 2020-09-15:
Just to say, my OctoDash was working perfectly - and now I get exact same behavior, it's hit or miss if it moves on from "starting".

This all happened when I updated OctoPrint to Python 3 - https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

I did a manual update on OctoDash and it's still the same.

dani190, 2020-09-16:
You know, it appears I forgot to mention that specifically. I am on Python 3 also but I started using Octodash after upgrading.

TheNeskik, 2020-09-18:
I get the same behavior.
I updated to Python 3 : Everythings worked good but i had some drag and drop issue
Updated to latest Octodash to solve this issue : Stuck on "started" + can't re-organize file on my pi (can't click on the red cross when i select some filter)

haplm, 2020-09-18:
The same problem - Octodash updated to Python 3, the screen is stuck in "Starting". Is there anything I can do to help debug this?

haplm, 2020-09-18:
I have just realised that this is a broader problem, as this is the situation that I see on my dashboard (which also says "Starting"):
[image]

haplm, 2020-09-18:
This seems to be the relevant part of the log:

`2020-09-18 09:07:43,414 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. LayerExpression valid. Start processing...
2020-09-18 09:07:43,562 - octoprint.filemanager.analysis - INFO - Starting analysis of local:3x3_cali_rl_0.2mm_PLA_MK3S_30m.gcode
2020-09-18 09:07:43,655 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: 3x3_cali_rl_0.2mm_PLA_MK3S_30m.gcode, owner: _api, user: _api
2020-09-18 09:07:43,686 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2020-09-18 09:07:43,693 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: 3x3_cali_rl_0.2mm_PLA_MK3S_30m.gcode, owner: _api, user: _api
2020-09-18 09:07:43,753 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-09-18 09:07:43,759 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2020-09-18 09:07:43,761 - octoprint.util.comm - ERROR - Error while running callback of QueueMarker
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 5038, in run
    self.callback()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 1194, in <lambda>
    self.sendCommand(SendQueueMarker(lambda: self._changeState(self.STATE_PRINTING)), part_of_job=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 697, in _changeState
    self._callback.on_comm_state_change(newState)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1116, in on_comm_state_change
    self._analysisQueue.pause() # only analyse files while idle
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/analysis.py", line 106, in pause
    q.pause()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/analysis.py", line 218, in pause
    self._do_abort()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 207, in _do_abort
    if _allow_analysis(self._plugin._printer, self._plugin._settings):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 188, in _allow_analysis
    if thermostat['target'] < 30:
TypeError: '<' not supported between instances of 'NoneType' and 'int'
2020-09-18 09:07:43,813 - octoprint.plugins.DisplayLayerProgress - INFO - Store layer count in MetaFile
2020-09-18 09:07:43,880 - octoprint.plugins.DisplayLayerProgress - INFO - Read total height from MetaFile
2020-09-18 09:07:43,889 - octoprint.plugins.DisplayLayerProgress - INFO - Total height not found in MetaFile
`

haplm, 2020-09-18:
And the corresponding issue in PrintTimeGenius: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/210

TheNeskik, 2020-09-18:
So it is not an Octodash issue.
We need to know if everybody with this issue have PTG.
A good starting point is to disable this plugin and print some files. I can not print now but I will give it a try later.

antonyburden, 2020-09-18:
I can confirm I am running PTG, I will disable it now and do some more tests.

UnchartedBull, 2020-09-18:
Thanks for getting to the bottom of this! :) Hopefully PTG will provide a fix soon. Once the status is set correctly, OctoDash should work fine again.

dani190, 2020-09-18:
Interesting, I have PTG also... Just uninstalling it now and will try and report back.

Thanks everyone!

UnchartedBull, 2020-09-21:
With PTG disabled OctoDash won't loose any features, just the estimated time remaining will be not so accurate anymore. PTG also just released an update, so maybe everything will work again :). Closing this here for now. If the issue comes back up, please reopen.

closed 2020-09-21 by UnchartedBull