UnchartedBull/OctoDash #1007: Compatibility issue with Bigtreetech PITFT50

3dbret, 2020-09-19:
**Describe the bug**

Strange behaviour with the BTT PITFT50.  When touched the screen will start multi-selecting/multi-touching many menu items (like 2 or 3 at a time) and the mouse cursor jumps all over. Sometimes on boot the screen behaviour is normal, sometimes it is not.  It can always be reproduced by waiting a while or by printing and cancelling a print.   

I have tried different screen polling times, calibration, updating things as per the wiki, but the same behaviour happens.  With the OctoBTT app from Bigtreetech it does not have this behaviour ever.  It would be great to get OctoDash working on this as their software leaves a lot to be desired.

**General Information:**

 - Raspberry Pi 4b 4gb.
 - OctoDash v2.0
 - Linux octopi 5.4.65-v7l+ #1341 SMP Tue Sep 15 13:13:32 BST 2020 armv7l GNU/Linux

3dbret, 2020-09-20:
After more testing it does not seem to be a problem with the touchscreen, there is something going more in general.  If you disconnect the USB cable the behaviour goes back to normal, until it happens again - so something to do with how it is handling data coming from the printer.  Again, I don't see this behaviour with the OctoBTT app, so not sure what is causing it.

Here is a video of the behaviour: https://imgur.com/a/kIufwtS

UnchartedBull, 2020-09-21:
Hm that is very weird indeed. I had something similar occur to me on the official Raspberry Display. Learned that these are called GhostTouches and probably are due to some weirdness going on with the Power Supply. I first tried adding a capacitor between the positive and minus of the screen power cables which made things better, but they still occurred from time to time. Switching out the power supply did fix it completely for me (none have occurred in the last 2 months).

Could you maybe try changing your Power Brick to the biggest / best quality one you have? If that doesn't work please report back so we can further investigate this.

3dbret, 2020-09-21:
> Hm that is very weird indeed. I had something similar occur to me on the official Raspberry Display. Learned that these are called GhostTouches and probably are due to some weirdness going on with the Power Supply. I first tried adding a capacitor between the positive and ground of the screen power cables which made things better, but they still occurred from time to time. Switching out the power supply did fix it completely for me (none have occurred in the last 2 months).
> 
> Could you maybe try changing your Power Brick to the biggest / best quality one you have? If that doesn't work please report back so we can further investigate this.

It might not be the power that is causing it because I just swapped out the original 5.1v 3amp Pi power adapter for a RavPower USB brick 5v/3a and it didn't help (I don't have anything else available at the moment).   

I can fix it by unplugging the USB to the printer, hitting connect from the sleep screen and plugging the USB back in....then it goes back to normal but I can make it happen again quickly by printing/cancelling a print.

Sometimes when I reboot it starts and connects and acts in the strange ghost touch way, and I have to reboot or unplug/plug the printer USB cable again to fix it.   

I tried disconnecting my HQ Picam but that made no difference either, everything else is pretty standard.

To me the cause seems to be the USB communication to the printer because it happens randomly/sometimes on boot, sometimes not, and I can always fix it by unplugging the USB to the printer and re-connecting.....(until it starts to happen again).  I'm running the latest firmware for the Raspberry Pi, maybe they messed something up.   That being said it doesn't do it in the OctoBTT app.

Sekisback, 2020-09-24:
I installed my BTT PITFT50 yesterday, without any problems. I run Octoprint 1.4.2 without Python3 on a RPi 3B+ 
Powered from the PSU with a BuckConverter.
Everything works as expected, so I guess it has nothing todo with Octodash

3dbret, 2020-09-24:
I'm using a Pi4b 4GB though, and a bunch of Octoprint plugins which maybe
is another issue.  I already tried 3 USB cables, no difference.  Everything
is a fresh install.  I have tried 2 power supplies including the official
Raspberry Pi 4 one which should be ample at 3amps 5.1v.

TheNeskik, 2020-09-25:
> I can fix it by unplugging the USB to the printer, hitting connect from the sleep screen and plugging the USB back in....then it goes back to normal but I can make it happen again quickly by printing/cancelling a print.

Does your screen working as expected without your printer connected ? (ie. without USB plugged in ?)

3dbret, 2020-09-25:
yes, it also works fine in the OctoBTT app from BTT.

UnchartedBull, 2020-09-28:
What mainboard are you using for your printer? Another option could be that the Printers mainboard is putting in some weird voltage which causes the 5V line to freak out resulting in these Ghost touches (don't know whether that is true / technically possible). Did you try another USB port on the Pi by any chance yet? I think everything else has been tested yet ...

3dbret, 2020-09-28:
actually I had thought of that over the weekend so I disabled the 5v pin,
and is still does it unfortunately - must be something to do with the pi4
or weird usb data - It is an Artillery X1 / Maker Labs Gen L  8bit board,
running a custom firmware though from 3dnexus.

UnchartedBull, 2020-09-28:
Maybe try the virtual printer thing, if the issues don't occur with the virtual printer then there is really something weird going on with the USB connection.

strangecalling, 2020-10-09:
I don’t have the issue on an Ender 3 with BTT e3 mini control board.

Suspect EMI is causing the issue. Perhaps try a ferrite ring on the usb cable to the printer.

closed 2020-10-21 by UnchartedBull

mptrs, 2020-12-14:
See this is closed, but I would like to share what I found cause I was having the same issue. When I disconnect my fan, which was connected on the GPIO, the screen is working just fine. So looks like a power issue/conflict, I'm using the official Raspberry Pi powerplug. (for the raspi 4)

3dbret, 2021-01-15:
Still does it for me with the fan disconnected.

guiguijke, 2021-01-15:
Hello , I have exactly the same issue , tried with Pi3+ & Pi4 - powersupply 5v 3A - board skr 1.4 pro

UnchartedBull, 2021-01-23:
I had a similar issue with the original Pi display, for me powering the display with a separate 5V PSU did the trick (not powering via the GPIOs). There are also multiple threads available online on why this happening with the official display. I don't know if the PITFT50 does support separate power input, but if it does it might fix your issues.

bigrizzo, 2021-05-23:
I see the issue is closed but I'm experiencing this issue with the PiTFT50 and Octodash. Lots of phantom touches. I'm running a Tevo Tornado with a GenL 1.4 board and 5v 3A power supply. It worked ok with OctoBTT so might go back to that until this gets sorted. I'm not keen on trying to power the display with a separate power supply for now as I have everything mounted and stuck in place. A shame as I really like Octodash.

jneilliii, 2021-05-24:
I use the PiTFT50 without issue. Did you start from a fresh image when transiting to octodash from btt?

bigrizzo, 2021-05-24:
> I use the PiTFT50 without issue. Did you start from a fresh image when transiting to octodash from btt?

No, I didn't start from a fresh image. I started with a fresh Octoprint then installed OctoBTT, deleted OctoBTT then installed Octodash.

3dbret, 2021-05-24:
still have the issue on mine too.

Salvogi, 2021-07-23:
I have a Pi4 with BTT Pi TFT50 V1.0, when I connect the display to the Raspberry, the 2.4GHz WiFi no longer works with both OctoPrint and Raspbian.
With Octoprint, when I connect the display to the Raspberry after a few minutes the camera stops working.
As a power supply I use a 5A XL4015 step-down set at 5.1V connected to a 12V 20A power supply.
I think the problems only relate to the Pi4 with DSI displays.
Unfortunately I think the BTT PI TFT50 V1.0 does not have the possibility of external power supply.

badbod, 2021-08-17:
Assuming PI PSU provides required current.

Many printers have issues with ghost touch, even 12864 display will exhibit phantom button press. The fault is usually PSU noise due to bad grounding.  in the case of step downs like  XL4015 make sure to add more smoothing. Cheap ebay, off the shelf, generally skimp on the smoothing and are very noisy. Add smoothing at the drop down output, dont let the noise out onto the wire.

Due to 3d printers industrial heritage they treat 0V and ground differently to consumer electronics so when placed in a consumer environment you do have to 'make them safe'. For your safety you will have to check and add ground to your printer chassis as it will likely not be already done (properly).

  Ensure printer PSU casing is properly earthed and is electrically connected to printer chassis. Most of a 'certain brand' are not.
  Recommended is adding grounding wires to the stepper motors metal case.
  Add a ground spike if needed (or desired, no harm adding more safety if you can).

  Disconnect the 5V wire from the USB. The controller board is likely to be unreferenced to ground or the 0V of attached devices so 
    A) Anything on here will also not be earthed.
    B)  More importantly, not referenced to external equipment 0V

  At some point in the chain , usually PC, it will get earthed and both 0V and ground will be connected. To avoid issues best to directly add earth to the printers USB port casing. This may also connect your printers 0V to earth, but that's going to happen anyway and is only a good thing to prevent sparks and damaged IC's when plugging in devices that are referenced to both 0V and earth, like your PC.

   Try to shield any ribbon cables or move them away from power supplies, like your printer PSU and  XL4015 drop downs.
Some drop downs (cheap ebay crap) do not like 0V being referenced to ground. Ensure you do not use them. your input 0V will be connected to to output 0V (and ground) at some point, in some piece of equipment, whether you like it or not.  

With good grounding, earth and smoothing all phantom clicks should be gone.