UnchartedBull/OctoDash #1009: Add Nightly Builds of OctoDas

o5, 2020-09-24:
**Is your feature request related to a problem? Please describe.**
I would like to test current master version with #873

**Describe the solution you'd like**
http://gnethomelinux.com/OctoPi/nightly/

**Additional context**
I looked at [CONTRIBUTING](https://github.com/UnchartedBull/OctoDash/blob/master/CONTRIBUTING.md) guide to build locally but without success on Mac with base environment (node, npm). Also Docker build would be cool if possible.

TheNeskik, 2020-09-25:
I agree with you, it could be a good things to add.

I also tried to build locally. In Windows it is impossible since electron doesn't work on windows
On Ubuntu I managed to build the 3 deb file but after installing the package on my Pi3, it turned on a black screen after Ratpoison start. (It seems that my ubuntu is on amd64 and some dependencies are not compatible with armv7) but that's an other issue and I need to go deeper to understand everythings :)

M1dn1ghtN1nj4, 2020-10-01:
@UnchartedBull is welcome to remove the link if he wants, but I have started a section on my website specifically for semi-nightly builds, as I am doing testing and customization as well.  Let me know if they builds fail to work properly.  Of course, there are some bugs, but as long as it doesn't fail to boot up or anything major, it should be good.

https://sites.google.com/view/m1dn1ght-3d/downloads/unofficial-octodash-builds

TheNeskik, 2020-10-01:
Hi, nice work, On my fork, i managed to do some nightly build too.

For me, every build doesn't work after the commit 6458cc6ec802882718979c0ca1af35eb0c279828
This commit use mdns and require to be re-build in order to work on raspberry. I didn't managed to do it, i will open a new issue for this

UnchartedBull, 2020-10-01:
@M1dn1ghtN1nj4 thanks! Would you mind if I link that site in the README / wiki? Most recent build won't install until https://github.com/UnchartedBull/OctoDash/tree/fix/text-wrap has been merged as described in https://github.com/UnchartedBull/OctoDash/issues/1033#issuecomment-702033209.

M1dn1ghtN1nj4, 2020-10-08:
Sure!  I'm still trying to set up a one line command that will install my build like you have.  I have limited time, but doing more on that tonight.

M1dn1ghtN1nj4, 2020-10-08:
@UnchartedBull I created a new repo on my account, and will be posting the builds over there.  I now have it linked to a modified install.sh script to make installing my builds easier.  I still need to update my website to reflect that.  So you'll be able to link to either my GIT page or my website.  Or both if you prefer.

M1dn1ghtN1nj4, 2020-10-08:
Ok, got the install script working properly now.  Looks like everything is good.  Currently the only issue I have is that after hitting SAVE in the settings menu, I get a blank screen, and the system is unresponsive.  Other than that, it looks good.

TheNeskik, 2020-10-08:
I'm having the same issue with the save button, I'm looking for this error.

I can say that if we look in ssh and try to run octodash, it returns 'Segmentation fault'. I'm wondering if this can be linked to the performance issue.

M1dn1ghtN1nj4, 2020-10-08:
It has been uncharacteristically slow recently.  I'm still working through
the code.  I've never dealt with angular or electron before.

TheNeskik, 2020-10-08:
Yes, with some tests, I can say that octodash is slower when idle than when printing.
When i print, my screen is very responsive.
In my opinion, it became slow when Prettier plugin was added. But that's strange because i should not occure some performance issue.
When UnchartedBull made some change to prevent drag element, he changes the global css file with 
*::After et *::before and maybe that is the cause of the issue. I read an article saying that those flag in css can slow the loading of the web page. Maybe that's what we are experiences with our Pi with less ressources than a normal computer ?
I'll look deeper this weekend for this issue

M1dn1ghtN1nj4, 2020-10-08:
I am running this on a Pi4 4GB, and 16GB C10, U1, V10, A1 microsd card.  I
also have a couple extra 3B+s in the drawer.

If you have a change to test, I'll see what each one does.

UnchartedBull, 2020-10-08:
I'll have a look at that as well once I have some spare time. Feel free to create a PR if you've found a fix :). Also the *::before and *::after aren't doing anything I think (they were mentioned in the article to prevent drag events, so I thought it can't hurt). Just remove them if you think that fixes the issue.

UnchartedBull, 2020-10-09:
Closing this here for now, since the nightly builds are available thanks to @M1dn1ghtN1nj4. I'll update the ReadME soonish. The slow interface will be tracked in #1062. Thanks again!

closed 2020-10-09 by UnchartedBull

UnchartedBull, 2020-10-11:
Just a quick update: The issue was the upgrade to Electron 10, after I downgraded to 9 everything runs smoothly again. Still don't know why it is acting up on Electron 10, but that will be discussed in the other issue. Release 2.1 should be ready soon :)

M1dn1ghtN1nj4, 2020-10-11:
Awesome.  When you get the commit to bump it back down to 9.  I'll do
another build, and see how things work.  I was going to start a build
tonight, then saw your email.

UnchartedBull, 2020-10-11:
https://github.com/UnchartedBull/OctoDash/commit/6d08f3c5971242d07d2ab63c065098fd86ab93bb fresh and hot 👍

M1dn1ghtN1nj4, 2020-10-11:
> 6d08f3c

Perfect.  Starting builds now.  Do you want to wait on the one pull request
that's open?  Just build it as it is right now?

UnchartedBull, 2020-10-11:
The other PR is unrelated, master should be fine now.

M1dn1ghtN1nj4, 2020-10-12:
OK, so I'm still trying to narrow this down.  I install a clean copy of
OctoPrint.  Do a full-upgrade.  Install OctoDash using the most recent
build and install.sh (only changed to point to my deb file).

It never boots to Ratpoison or OctoDash.  I tried going over all the manual
installer instructions, and even checking file permissions.  Tried with the
minimum pluginsNothing is working so far.  Just boots to terminal.  I did
see it says this:

*/hom/pi/.xinitrc: 8: /home/pi/.xinitrc: octodash: not found*

M1dn1ghtN1nj4, 2020-10-12:
Hold that thought.  I think I found it.

M1dn1ghtN1nj4, 2020-10-12:
Nope, still broken.  Going to keep trying, unless you have some suggestions

M1dn1ghtN1nj4, 2020-10-12:
Yeah, if I manually do a wget on the file location, and store it as
octodash.deb, then dpkg on it, it installs just fine.  Something isn't
working in the install.sh script.

M1dn1ghtN1nj4, 2020-10-12:
Is there a better way we can chat?  Discord or something?

M1dn1ghtN1nj4, 2020-10-12:
It's all good now.  I had to fight with the releaseURL.

UnchartedBull, 2020-10-12:
just created - https://discord.gg/gTasZTz :)