UnchartedBull/OctoDash #1013: Cannot start OctoDash

3DFans, 2020-09-24:
Hey, 
[image]
I've now installed everything twice according to the instructions from Cris, but the same error always comes up and a start is therefore not possible. I use a Raspi 3b with the Waveshare 5 inch display.

the1snm, 2020-09-25:
First time I installed it I got the same error and if you look at  the Xorg.0.log it likely shows you don't have fbturbo installed. Here are the steps I did before running the installer for OctoDash on a fresh Octopi install:

`#Raspberry Pi4 4gb
#UCTRONICS 3.5" SPI Display U6111
#Installed latest Octoprint to MicroSD
#Using default raspberry password through this after each sudo as needed.

#Putty login in pi:raspberry
sudo apt-get update
password: raspberry
sudo apt-get upgrade
Y #Yes
Q #Quit the changelogs
sudo reboot now

#Putty login in pi:raspberry
#This next section were the drivers for my SPI display relpace this with yours.
git clone https://github.com/UCTRONICS/UCTRONICS_HSLCD35.git
cd UCTRONICS_HSLCD35/Octoprint
sudo chmod +x UCTRONICS_HSLCD35_SHOW
sudo ./UCTRONICS_HSLCD35_SHOW
Forced Reboot

#This is the step you're likely missing...
sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils
Y #Yes
sudo apt-get install git build-essential xorg-dev xutils-dev x11proto-dri2-dev
Y #Yes
sudo apt-get install libltdl-dev libtool automake libdrm-dev
Y #Yes
git clone https://github.com/ssvb/xf86-video-fbturbo.git

cd xf86-video-fbturbo
autoreconf -vi
./configure --prefix=/usr
make
sudo make install
sudo cp xorg.conf /etc/X11/xorg.conf`

NOW install OctoDash and if nothing displays right you might need to do the following. I needed to because my display is a hat and the drivers pointed to the wrong output.

`sudo nano /etc/X11/xorg.conf.d/99-fbdev.conf
Change: Option "fbdev" "/dev/fb1" to Option "fbdev" "/dev/fb0"
Ctrl-O, enter, Ctrl-X
sudo reboot now`

Screen rotation, touch screen axis's or touch rotation off is a different set of steps as well.

Hope this helps someone.

UnchartedBull, 2020-09-28:
Thanks @the1snm! I formatted your response a little so the code and commands are easier to read :)

@3DFans please report back whether everything is working now.

3DFans, 2020-09-28:
> the1snm!

It works with the instructions from the1snm. ;)

closed 2020-10-05 by UnchartedBull

theshortest, 2021-01-17:
Any help on this. I've been fighting this for days. My UCTRONICS LCD35 SPI works until I install Octodash. I've followed this and many other ways as well. This got me to see Octodash but I was completely unable to get it calibrated. I'm at a loss!

UnchartedBull, 2021-01-23:
Did you run the calibration script after installing?

theshortest, 2021-01-23:
I installed a desktop, ran the calibration figured out a matrix rotation and then transferred that config (in nano) to a new octo install so I didn't have the desktop bloating my system. I'm the dude who shared my touch screen install walk through on discord.

UnchartedBull, 2021-01-25:
I just had time to put your guide in the wiki: https://github.com/UnchartedBull/OctoDash/wiki/User-Guides#uctronics-lcd35-spi-install. Thanks again!

theshortest, 2021-01-26:
That's great. I noticed one thing I left out was to set console auto login
in raspi configuration.

jappyjan, 2021-03-03:
i dont have the file "/etc/X11/xorg.conf.d/99-calibration.conf"

what should i do?

theshortest, 2021-03-03:
If you do the walk through on the wiki it will create one in the process.
https://github.com/UnchartedBull/OctoDash/wiki/User-Guides

traviselam, 2021-08-11:
> I just had time to put your guide in the wiki: https://github.com/UnchartedBull/OctoDash/wiki/User-Guides#uctronics-lcd35-spi-install. Thanks again!

I just want to say thank you so much for posting this guide. I've been going in circles for about a day now trying to get this working and I finally got it! I wanted to use octodash because I didn't want to install a full desktop or browser on my pi just to get a touch screen up and working and this is perfect. Thank you!