UnchartedBull/OctoDash #1015: Working of [KILL] command

Protoncek, 2020-09-25:
What exactly does KILL (or SHUTDOWN) command do? I use it for shutting down my Pi when print is finished. Does it execute something like "sudo shutdown now" or anything else? 
Namely, today i experienced a second corruption of my installation. Appr. a week ago i've had a ton of errors when PI was booting and i had to reinstall all from scratch. OK, first time i blamed it on corrupted MicroSD card, so i installed all on SSD drive instead (i have Pi4, it boots from SSD now). 
But today i've had a second failure - although a different one (**/usr/bin/init: cannot execute binary file exec format error**). When i tried to reinstall octodash i've got an error (after finished donwload of package): 
dpkg: unrecoverable fatal error, aborting:  files list file for package 'hdparm' is missing final newline**

Luckily a few days ago i've made an image, so i solved pretty quickly. Maybe it's not connected with OctoDash, but i'm wondering what's wrong all of the sudden? I've had zero problems for quite a while. Since i changed media it can't be that...

UnchartedBull, 2020-09-28:
That shouldn't happen. I didn't have any issues with the microSD I'm using for little over a year now. KILL just calls SHUTDOWN internally, which then tells OctoPrint to shutdown. So it will use whatever OctoPrint uses when you shut it down via the WebUI.

Personally I just kill the power to Pi, which didn't cause any issues :) Let me know if you have more questions.

Protoncek, 2020-09-28:
I've been using KILL (or SHUTDOWN) command a while without any issues, too. That's why i find it very odd that suddenly these errors came out... Thanks for explanation, at least now i can be sure that this command can't be the issue. 
So next thing is Octoprint itself, or any of it's updates or plugins. When all goes wrong i get, beside tons of errors, message at the end: 
**- bash: /usr/bin/xinit: cannot execute binary file: Exec format error** 
and nothing's running, not server, not OctoDash. So far i didn't find any reasonable explanation for this error. i'll keep searching...

Thanks!

closed 2020-10-09 by UnchartedBull