UnchartedBull/OctoDash #1016: White Box when touching

scorer21, 2020-09-25:
When touching the screen I see a white box where I touch, but the button doesn't actually activate. The box can be dragged arround when holding the touchscreen. Click only works after severeal tries.

[image]

UnchartedBull, 2020-09-28:
Please provide more info (OS version, touchscreen, other libraries, fresh install?, ...)

scorer21, 2020-09-28:
- OctoPrint 1.4.2 
- Python 3.7.3 
- OctoPi 0.17.0
- Raspian Buster (Octoprint Image - relatively fresh install)
- Touchscreen:
[image]

I found a fix you provided on another forum for exact the same problem. Tried it out, fixed the problem but it slowed down the interface tremendously. Reverted back to the latest official version (which is fast and responsive).

TheNeskik, 2020-10-06:
What was your first solution ?
Are you using the latest official release of Octodash or a personnal build ?

scorer21, 2020-10-06:
I used the version @UnchartedBull provided in issue #838 . This solved the problem with the white box but unfortunately I then had the problem of a very unresponisve interface which was very laggy. I then reverted back to the latest official version.

UnchartedBull, 2020-10-09:
Closing this here now, since the white box is gone. I created a new issue for the sluggish interface: #1062.

closed 2020-10-09 by UnchartedBull