UnchartedBull/OctoDash #1017: Can't install octodash on ubuntu server

bepstein111, 2020-09-25:
**What doesn't work?**
Using the install script fails with `Package gir1.2-gnomekeyring-1.0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source`

**What did you already try?**
Just tried installing.

**General Information:**
 - Ryzen 2700X build with an olllllddd 8400GS
 - Ubuntu Server 20.04
 - OctoDash Version [latest]
 - OctoPrint Version [latest]

**Additional context**
I had octoscreen installed before, but I uninstalled/purged it before trying to install this.

carl1961, 2020-09-27:
i have same error on Debian 10. funny last week i installed it . been working on orange pi , so lots of reinstalling, testing. every where i read it has been removed

UnchartedBull, 2020-09-28:
Dependency has been removed. Please try again after the PR is merged.

PR #1028 (2020-09-28): update dependencies
Fixes #1017.
Closes #1019.
closed 2020-09-28 by UnchartedBull