UnchartedBull/OctoDash #1018: Will not install on PI4

maverickccie, 2020-09-25:
When trying to install on a PI4 with all current updates the install will not run. Get this:

pi@octopi:~$ bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)
pi@octopi:~$ 

**General Information:**
 - Hardware: PI 4 2gb
 - OctoPrint version : 1.4.2
 - OctoPi version    : 0.17.0

the1snm, 2020-09-25:
Make sure you don't have extra spaces and such in there as I just installed this 2 days ago on a Pi4 and it works fine, in fact it's printing right now.

maverickccie, 2020-09-25:
> Make sure you don't have extra spaces and such in there as I just installed this 2 days ago on a Pi4 and it works fine, in fact it's printing right now.

I am copying strait from github into the ssh session. made sure there is no extra spaces.

maverickccie, 2020-09-25:
Something must really be wrong because I just tried it on a Pi 3 and same thing  return to prompt right away.

Protoncek, 2020-09-26:
Hm... i just installed it with same copy/paste system and it worked on my Pi4. So,. i guess it's nothing wrong with link / command.

UnchartedBull, 2020-09-28:
There definitely is nothing wrong with the installer script as it is working like that for a few months now. 

What shell are you using? Can you please post the output of `tail -5 ./install.sh`. Are there any errors on the console?

maverickccie, 2020-09-29:
This is what I get (no errors):

pi@octopi:~$ tail -5 ./install.sh
list_input "Shall I reboot your Pi now?" yes_no reboot
echo "OctoDash has been successfully installed! :)"
if [ $reboot == 'yes' ]; then
    sudo reboot
fi
pi@octopi:~$ 

I  am able to install via the manual way but cannot run the script, so I know Octodash works but the install script does not.

UnchartedBull, 2020-10-09:
that script looks fine. there is probably something funky going on with your console. Since you got it working (although only via the manual way) and nobody else had the same issue I'm closing this here for now.

closed 2020-10-09 by UnchartedBull