UnchartedBull/OctoDash #1023: Percentage is diffrent.

bchilton9, 2020-09-26:
Im not sure if there is a setting im missing or not. Octodash shows about 15% less then the webUI (Octopod on my phone shows 10% less too). Im thinking it might be print time genius adds the pre heat into the progress done but im not sure. Is there a way to sync them to the same percent? I dont care if the pre heat time is in there or not i would just like my screens to be in sync. Thank you.

UnchartedBull, 2020-09-28:
OctoDash currently uses the file position for percentage calculations (default from OctoPrint). PTG somehow only replaces the numbers in the UI (now with time remaining) and not in the API, thus there could be a difference. I'll have a look and check whether I can get the PTG time somewhere :)

closed 2020-10-22 by UnchartedBull

KapJI, 2021-12-30:
For anyone who finds this issue:

You can install https://github.com/celogeek/OctoPrint-ProgressBasedOnTime plugin to make octoprint to report more accurate percentage.

PR #1112 (2020-10-22): Use OctoPrint completion over self-calculated percentage
Closes #1023.