UnchartedBull/OctoDash #1026: Large G-Code Print From SD Results in Error

JohnTRoth, 2020-09-27:
When printing a (very) large G-Code file from the SD card with a Raspberry Pi 3b, OctoDash displays "Can't retrieve job status" continuously, OctoPrint correctly displays the job in progress.

Steps to reproduce the behavior:
I grabbed the moon-lamp-thingi_1m.stl from https://www.prusaprinters.org/prints/23859-designer-lithophane-moon-lamp/files, sliced it at .15mm and got a resulting 207MB G-Code file. Stored it on the Prusa SD card, then had the Prusa print it. OctoPrint correctly displayed the resulting file, print status, time remaining, filename, etc - but OctoDash is continuously displaying "Can't retrieve job status".

I am on the latest version of firmware, slicer software, and OctoPrint (having just upgraded everything earlier in the day). I just switched to OctoDash yesterday though, so I can't say if this issue was present in earlier versions of anything or not. I do have an MMU, though this print is for a single material, so I don't expect that to play into the equation.

UnchartedBull, 2020-09-28:
I'll have a look at that once I'm back home. Might be a couple of days though

UnchartedBull, 2020-09-28:
Just saw that there is another issue for this already (although with less details - thanks for the link to the file.) Will be tracked over in #941.

closed 2020-09-28 by UnchartedBull