UnchartedBull/OctoDash #1029: Adafruit 3.5TFT always has the touch rotated a -90 degrees from the screen

rv10guy, 2020-09-29:
I am attempting to setup OctoDash on a Raspberry Pi4 with the Adafruit 3.5 TFT. I am installing the drivers following the guide here [(https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/overview)]. No matter what I do, the touchscreen ends up -90 degrees off from the touchscreen. So for example, if I want to press the upper right corner, I have to press the upper left corner and if I want to press the upper left corner I have to press the lower-left corner. It does not matter how I rotate the screen, the touch, and the display are always off by -90 degrees. I also tried the steps here [https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/faq#faq-2859093] and it did not help.

carl1961, 2020-09-29:
maybe this link will help
https://whitedome.com.au/re4son/topic/my-adafruit-3-5-setup-for-those-having-inverted-axis-out-of-the-box/

UnchartedBull, 2020-10-09:
Probably not a problem with OctoDash, but rather with your setup. Please reopen this issue if @carl1961 link doesn't work.

closed 2020-10-09 by UnchartedBull

sabarrett328, 2020-12-30:
@rv10guy did the link help with this issue?  I have the same thing going on. When I get home I was going to try it.

rv10guy, 2020-12-30:
No, I was never able to get it to work. I abandoned the screen.

KTheMan, 2021-01-16:
Copy-pasting my response from discord, since it's more searchable here:

So, I'm definitely sure the touch is working correctly out of the box and it is Adafruit's installer script missing something. You can run `ts_calibration`  then `ts_test` and everything is perfect upon testing but X11 (which is what I'm assuming OctoDash is using to display through) doesn't pick up on it because the installer script looks for lightdm to rotate touch on X11. Install lightdm through apt, run the installer script again, and then remove lightdm. Reboot. Voila!

I did also have lxde (without its dependencies) installed/uninstalled at the time as lightdm but I do not believe that contributed to this solution.

Tl;dr: I seem to have solved it! Install lightdm through apt, run the installer script again, and then remove lightdm. Reboot. Voila!

sabarrett328, 2021-01-24:
hhmmm... Thanks for the info.  I"ll have to try that.  However.... I'm not a Linux guru by any means.  I've stumbled around searching for info on lightdm, and I found the github page, however I don't see anything referring to install by apt.  Could you please point me somewhere I can learn more about doing this?

thanks

machavezg9, 2021-01-27:
If I may give my 0.2, after hours of trying it came down to doing everything in the following two links.
Follow the instructions [here](https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/easy-install-2) 

Note: all code below is just a copy from the original source. Double check for mistakes or run from original source.

Install your 3.5" tft
`cd ~`
`sudo pip3 install --upgrade adafruit-python-shell click==7.0`
`sudo apt-get install -y git`
`git clone https://github.com/adafruit/Raspberry-Pi-Installer-Scripts.git`
`cd Raspberry-Pi-Installer-Scripts`

I ran through the interactive installation:
'sudo python3 adafruit-pitft.py'
- Installed my PiTFT 3.5" 
- Selected [4]
- Selected [1] for 90 deg orientation
- No to console
- Yes to HMDI mirror
- Reboot

After Reboot make sure your Octodash is displayed and running in the (all touch for me was portrait but landscape gui orientation)

Now run the manual calibration [here](https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/touchscreen-install-and-calibrate#manual-calibration-2859019-17)
- Note the manual calibration might not be needed but I included this because I did run it a few times til I was able to run the touch screen test and draw accurately. (Might be able to skip this)

Want may have done all the magic.

Run X Calibration [here](https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/touchscreen-install-and-calibrate#x-calibration-2859025-23)

Install xcal per X Calibration linked above:
`sudo apt-get install libxaw7-dev libxxf86vm-dev libxaw7-dev libxft-dev`
`git clone https://github.com/KurtJacobson/xtcal`
`cd xtcal`
`make`

Reset your display:
`DISPLAY=:0.0 xinput set-prop "stmpe-ts" 'Coordinate Transformation Matrix' 1 0 0 0 1 0 0 0 1`

Set your display geometry (NOTE!! YES 480x320 for PITFT, followed the "720x480" and it did not work):
`DISPLAY=:0.0 xtcal/xtcal -geometry 480x320` 

Follow the on screen directions
Copy your new "Transformation Matrix Coordinates"

Navigate to
`cd /usr/share/X11/xorg.conf.d/`
List everything in drive
`ls`
If nothing with the name "20-calibration.conf" is listed, create this file
`sudo nano 20-calibration.conf`
Copy and Paste the code below (sourced from the xcal Adafruit link), make sure to replace the "TransformationMatrix" values with the ones previously recorded:
```
Section "InputClass"
        Identifier "STMPE Touchscreen Calibration"
        MatchProduct "stmpe"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        Option "TransformationMatrix" "-0.000087 1.094214 -0.028826 -1.091711 -0.004364 1.057821 0 0 1"
EndSection
```

Reboot your Pi
Hopefully it worked!

h3llb3nt, 2021-04-21:
did you actually run these steps yourself? the x calibration requires the use of PIXEL and with octopi there is no GUI desktop environment (as far as I know) so I cannot perform these calibrations...

BorkStick, 2021-05-06:
Tried the @KTheMan fix and it did the trick.

I was goin off the troubleshooting page on the wiki and was also confused by the "installation script" referred to. Unlike @h3llb3nt I was thinking it was referring to the octodash script, and spent way longer than I would have like to figure out it was referring to the adafruit install script 

Can we tweak the wiki section for this issue so it reads `adafruit installer script`, instead of `installer script`

```
Fix
Install lightdm through apt, run the adafruit installer script again, and then remove lightdm. Reboot. Thanks to @KTheMan
```