UnchartedBull/OctoDash #1033: Unable to start Octodash with mdns on Pi3

TheNeskik, 2020-10-01:
**Describe the bug**
Since 6458cc6ec802882718979c0ca1af35eb0c279828, i'm not able to build and launch Octodash on my Pi3.
I know that we need Zeroconf/libavahi-compat-libdnssd-dev. 

**To Reproduce**
Steps to reproduce the behavior:
1. Clone repository on clean ubuntu/macOS
2. Only on Ubuntu : install libavahi-compat-libdnssd-dev
3. npm install
4. npm run pack
5. get the armv7l deb file
6. On the pi : install libavahi-compat-libdnssd-dev
7. Install the armv7l deb file
8. reboot
9. Black screen on the pi
10. With SSH : try to launch octodash
11. Error about mdns with ELFCLASS or ELFCLASS64 depend on the build plateform

**Expected behavior**
Octodash launch on Raspberry

**General Information:**
 - Hardware : Raspberry Pi3 to run octodash + my computer to run builder
 - OS Info : Octopi 0.17.0 + On my computer as virtual machine : Ubuntu 20.04 and MacOS catalina
 - OctoDash Version  v2.0.0+latest commit
 - OctoPrint Version 1.4.2

**Additional context**
npm start is working correctly on ubuntu and MacOS as mdns is correctly build for the arch

We probably need to build and compile on raspberry but the pi3 doesn't enough ressources for this. Maybe a Pi4 can do the job...

UnchartedBull, 2020-10-01:
mdns also gave me a lot of headaches. Replaced it with bonjour already (which doesn't rely on gyp to be build and I think (not tested though) does not require libavahi-compat-libdnssd-dev. It's in this branch: https://github.com/UnchartedBull/OctoDash/tree/fix/text-wrap let me know if it works :)

TheNeskik, 2020-10-01:
Nice man !
I was able to build it and it starts like a charm. I don't know if it really works because my printer was already configure..
I will start some print now to see if text-wrap is solved too. If somebody would like to test it I made a pre-release in my personal fork :
https://github.com/TheNeskik/OctoDash/tree/v2.0.3-alpha

One more point, the screen is very long to respond as mentioned in this issue #838
I will look into this in the future

TheSin-, 2020-10-02:
you would need to pack on the arm you can't pack it on x86 and run it on arm as it's building static binaries.

TheNeskik, 2020-10-02:
Yes, I understood that if i pack on x86 or x64, the binaries are building for those arch. 
But I don't understand that we can't rebuild it for an other arch. That's why we use electron, to do some cross-compilation.

I'm sorry if I miss some information that can be obvious.

I will re-test with mdns when I will have a pi 4 as I can't build it on Pi3. Did you manage to build it on Pi3 ?

TheSin-, 2020-10-02:
I did it on Pi4, but I"ll test it on Pi3 tonight I have one some place ;). But the Pi4 binary should work on Pi3 as they are both armhf

That being said the changes that UnchartedBull made should negate the need to build a static version of mdns I believe.

TheNeskik, 2020-10-06:
This issue is now solved.
The long time for screen response is link to another issue.
Thanks all for your help

closed 2020-10-06 by TheNeskik