UnchartedBull/OctoDash #1039: "No screens found" İssue

furkanbr, 2020-10-03:
**What doesn't work?**
[image]
Octodash is not working , giving me this error.

**What did you already try?**
I installed octopi and octodash few times from 0 , check the drivers , raspi-config , and even tried with RPi zero but it didnt work.

**General Information:**
 - Hardware ; RPi 4 4GB Model , Waveshare 3.5inch RPi LCD (A), 480x320 
https://www.waveshare.com/3.5inch-RPi-LCD-A.htm
 - OS Info ; OctoPi Buster Lite 0.17
 - OctoDash Version ; Newest
 - OctoPrint Version ; Newest

**Additional context**
I added error that i get , i set my PRi to Autologin Console . When i turn on my Pi , octodash not showing even a second , just giving an error. Please help me to make it work. Thanks , have a good day.

furkanbr, 2020-10-05:
Does anyone know how to solve ? im still getting this issue and didnt find how to solve.

TheNeskik, 2020-10-06:
It doesn't look like an Octodash issue but an issue with xorg.
I can see here 2 problem : 
- No screen found -> Xorg configuration isn't pointing the right screen.
- Connection refused -> your pi user doesn't have the access to launch X server.

If you are on a fresh install of Octopi you need to install the touch driver as describe here first : 
https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)#Install_the_touch_driver

Secondly, you need to install Octodash with the script :)

If you already did the first point and you have the same issue, you will need to grant some permissions.

furkanbr, 2020-10-06:
@TheNeskik Thanks for your answer !! i solved my problem with this link , if anyone having same problem it can help ;
https://raspberrypi.stackexchange.com/questions/100169/xorg-not-working-after-upgrading-to-buster/101085#101085?newreg=f7832854b4fe47a29a1597dafdb93ff6

closed 2020-10-09 by UnchartedBull

kekko7072, 2021-02-05:
I having the same error an I'm not able to solve it
This is the log file of the error. I tried everything also installing the desktop but nothing works. My LCD is correctly installed infact if I try desktop version of Raspbian it works perfectly the lcd and the touch scree. (THIS IS LINK TO WIKI OF LCD https://www.waveshare.com/wiki/5inch_HDMI_LCD)

PLS HELP ME!!

[    52.429] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    52.433] Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
[    52.434] Current Operating System: Linux octopi 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l
[    52.434] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
[    52.435] Build Date: 15 January 2021  02:03:57PM
[    52.435] xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support) 
[    52.436] Current version of pixman: 0.36.0
[    52.436] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    52.436] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    52.438] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Feb  5 22:22:03 2021
[    52.482] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    52.507] (==) No Layout section.  Using the first Screen section.
[    52.508] (==) No screen section available. Using defaults.
[    52.508] (**) |-->Screen "Default Screen Section" (0)
[    52.508] (**) |   |-->Monitor "<default monitor>"
[    52.544] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    52.545] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    52.545] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    52.545] (==) Automatically adding devices
[    52.545] (==) Automatically enabling devices
[    52.545] (==) Automatically adding GPU devices
[    52.546] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    52.554] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    52.554] 	Entry deleted from font path.
[    52.555] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    52.555] 	Entry deleted from font path.
[    52.555] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    52.555] 	Entry deleted from font path.
[    52.556] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    52.556] 	Entry deleted from font path.
[    52.556] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    52.556] 	Entry deleted from font path.
[    52.556] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    52.557] 	Entry deleted from font path.
[    52.557] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    52.557] (==) ModulePath set to "/usr/lib/xorg/modules"
[    52.557] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    52.557] (II) Loader magic: 0x1fdf80
[    52.558] (II) Module ABI versions:
[    52.558] 	X.Org ANSI C Emulation: 0.4
[    52.558] 	X.Org Video Driver: 24.0
[    52.558] 	X.Org XInput driver : 24.1
[    52.558] 	X.Org Server Extension : 10.0
[    52.602] (++) using VT number 7

[    52.603] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    52.604] (II) no primary bus or device found
[    52.605] (II) LoadModule: "glx"
[    52.614] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    52.774] (II) Module glx: vendor="X.Org Foundation"
[    52.775] 	compiled for 1.20.4, module version = 1.0.0
[    52.775] 	ABI class: X.Org Server Extension, version 10.0
[    52.775] (II) LoadModule: "fbturbo"
[    52.790] (II) Loading /usr/lib/xorg/modules/drivers/fbturbo_drv.so
[    52.800] (II) Module fbturbo: vendor="X.Org Foundation"
[... 176 lines cut ...]
[    63.939] (**) Option "config_info" "udev:/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1A2C:2124.0003/input/input3/event3"
[    63.939] (II) XINPUT: Adding extended input device "SEM USB Keyboard Consumer Control" (type: KEYBOARD, id 9)
[    63.939] (**) Option "xkb_rules" "evdev"
[    63.970] (**) Option "xkb_model" "pc105"
[    63.970] (**) Option "xkb_layout" "gb"
[    63.973] (II) evdev: SEM USB Keyboard Consumer Control: initialized for relative axes.
[    63.973] (WW) evdev: SEM USB Keyboard Consumer Control: ignoring absolute axes.
[    63.976] (**) SEM USB Keyboard Consumer Control: (accel) keeping acceleration scheme 1
[    63.976] (**) SEM USB Keyboard Consumer Control: (accel) acceleration profile 0
[    63.977] (**) SEM USB Keyboard Consumer Control: (accel) acceleration factor: 2.000
[    63.977] (**) SEM USB Keyboard Consumer Control: (accel) acceleration threshold: 4
[    64.054] (II) config/udev: Adding input device SEM USB Keyboard System Control (/dev/input/event4)
[... cut ...]
[    64.057] (--) evdev: SEM USB Keyboard System Control: Found keys
[    64.057] (II) evdev: SEM USB Keyboard System Control: Configuring as keyboard
[    64.057] (**) Option "config_info" "udev:/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1A2C:2124.0003/input/input4/event4"
[    64.058] (II) XINPUT: Adding extended input device "SEM USB Keyboard System Control" (type: KEYBOARD, id 10)
[    64.058] (**) Option "xkb_rules" "evdev"
[    64.058] (**) Option "xkb_model" "pc105"
[    64.058] (**) Option "xkb_layout" "gb"