UnchartedBull/OctoDash #1040: Keep getting error "Can't retrive layer progress!"

chilipeppr, 2020-10-05:
The bug I'm seeing is this:

[image]

I just installed this on a Raspi4 tonight. Ran the standard install script. The app is working fine and in fact when it first came up I was not seeing this error. After a reboot I'm seeing this.

Here is the other error it shows. It toggles back and forth between this and "Can't retrieve printer status"
[image]

**General Information:**
 - Raspi4
 - OctoPrint version : 1.4.2
 - OctoPi version    : 0.17.0

UnchartedBull, 2020-10-05:
Your URL isn't right. Please double check (either via UI or config file). If you don't know what is wrong with the URL please post it here.

chilipeppr, 2020-10-05:
I have it set to localhost without any port specified as I'm running OctoPrint on port 80 like the defaut SD card install of OctoPrint runs on.

Here's my current setting. Keep in mind I connect to OctoPrint on this Raspi just fine.
[image]

Even when I try the direct IP, which is what I type into my browser from my laptop to get to OctoPrint, that fails the same way:
[image]

Even tried doing loopback IP, but fails same way.
[image]

Here I am connected in my browser just fine to 10.0.0.202.
[image]

TheNeskik, 2020-10-06:
I'm running the same configuration as you, and I need to put it on port 5000. Give it a try :)
If it doesn't work, you can try to put it on port 80. 
In your pictures we don't see any port.

chilipeppr, 2020-10-06:
Yeah, that worked. Wow, I could have sworn I tried that at the beginning during the setup wizard and it failed until I finally thought to delete the port number thinking it would presume port 80, which it did, and then all was working. Perhaps it was always using port 5000 in the background and after reboot it finally started to use the empty value and that's why it failed?

Ok, thank you. 

[image]

UnchartedBull, 2020-10-06:
Great that you got it working! The setup will be simplified with the next release, so you don't need to worry about ips and ports anymore :)

closed 2020-10-06 by UnchartedBull

berteding, 2020-11-30:
I have the same problem. 
I tried to set the port to 5000, now i get Connection to Octoprint timed out.
And I can no longer access the settings because it keeps trying to connect end then times out.
Is there a configuration file where the settings are stored?

UnchartedBull, 2020-11-30:
https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#manually-adjusting-the-settings

nicwilson58, 2020-12-08:
> I'm running the same configuration as you, and I need to put it on port 5000. Give it a try :)
> If it doesn't work, you can try to put it on port 80.
> In your pictures we don't see any port.

I have this exact issue, I cannot change anything on the Pi screen as this error box keeps popping up over the top of the UI making it impossible to enter anything,  where else can I try setting these changes?  i looked at the setup link above but could not find anything there either?

berteding, 2020-12-08:
It must be 127.0.0.1:5000, so changing only the port-number does not work and causes a hangup.

Cordero808, 2021-02-27:
I am still getting this error and i have the address and port set

JeremyProffitt, 2021-03-21:
This seems to be undocumented, but DisplayLayerProgress Plugin MUST be installed.

Cordero808, 2021-03-21:
Sorry issue is resolved after I reinstalled.  Thankyou!!