UnchartedBull/OctoDash #1049: Octodash on a external Pi with a wifi connection.

RedSquirrell, 2020-10-06:
Hello everyone,

Im am currently looking for a easy way to access my Octopi interface. I have a Raspberry Pi running Octoprint connected to my 3D printer. And i have another Raspberry Pi with a touchscreen connected to it. Is it possible to run octodash on a the pi connected to the touchscreen, and then control my octoprint instance on the other Pi with something like a wifi connection?

-Bart

TheNeskik, 2020-10-06:
Hi,

Yes, when you run octodash, you can simply put the address of your Octoprint installation.

RedSquirrell, 2020-10-06:
What do i do when i get "Please specify OctoPrints full virtualenv path manually (no trailing slash)" while installing Octodash?

closed 2020-10-06 by RedSquirrell

TheNeskik, 2020-10-07:
In fact, the install script is working great if you have octoprint on the same Raspberry.

If you want to install it on other raspberry, you simply need to install everything manually.
Don't forget to install Display Layer Progress plugin in Octoprint, that is mandatory.

You can try to use this install script tha tis the same as the one provided by UnchartedBull but without some check about Octoprint. I don't if it works great but you can try and give me the result
`
bash <(wget -qO- https://github.com/TheNeskik/OctoDash/raw/TheNeskik-patch-1/scripts/install_remote.sh)
`

Even if you use this script, please don't forget to install Display Layer PRogress in Octoprint

frakki1986, 2022-02-22:
Hi there,
im struggeling to get the install going. I've running Octoprint on an Proxmox VM (192.16x.x.35) now i'm trying to run Octodash an a raspberry pi but cant get the correct Path for Octoprint.

Can someone lead me the way?

UnchartedBull, 2022-02-22:
Just insert the IP you're using to connect the OctoPrint UI and everything should work.

frakki1986, 2022-02-22:
> Just insert the IP you're using to connect the OctoPrint UI and everything should work.

@UnchartedBull thx for your answer. I feel pretty damn stupid because reading your instructions carefully would have made things so easy.

"If your OctoPrint instance is running on a different machine just type **_-_** in the following prompt."

One **_-_** can bring you hours of fun :)