UnchartedBull/OctoDash #1050: Filament Loading/Unloading doesn't work correctly

davide-molino, 2020-10-06:
I've modified the config.json file to put the right feed lenght and the desired speed. The config is accepted because the values change in the dashboard... but the speed remain the same. I want to load/unload with an higher speed but also with the dashboard displaying, for example, 100 mm/s it always works at 30 mm/s. How can I fix this issue?

closed 2020-10-07 by davide-molino

UnchartedBull, 2020-10-07:
I guess you figured out by yourself that you need to update your printer profile :). I added a section for that to the troubleshooting guide: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#filament-change-is-slower-than-specified-in-configuration and updated the explanation a little to make this more clear: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#feed-speed

davide-molino, 2020-10-07:
exactly... problem solved. Thank you for answering my question... Octodash is adorable