UnchartedBull/OctoDash #1052: Preheat Button shows error 405 if the thumbnail size is higher than 200*200

davide-molino, 2020-10-06:
Reading the new website I've decided to set the resolution of the thumbnail to 256*256. after saving and uploading the new gcode I've noticed that the preheat button doesn't work neither from octodash or the web interface... so I've tried to scale the thumbnail down and it seems that with 200*200 it will work. What can cause this issue?

UnchartedBull, 2020-10-09:
That certainly is really really weird. I'll investigate this :)

davide-molino, 2020-10-10:
If you need more material I can repeat my experience

UnchartedBull, 2020-10-15:
I just tested this with a 320x320 thumbnail and everything works as expected. Could you maybe provide the file so I can test it with the exact same file?

I'm thinking that this is probably a problem with the preheat plugin itself - PrusaSlicer is appending the image as a base64 string to the top of the GCOde, the plugin might not be able to find the temperature (maybe it online searches the first 800 lines of GCode, which would work up to a certain resolution). If I have the file I can further investigate this :)

UnchartedBull, 2020-10-15:
I just checked the API code for the preheat plugin and the 405 indicates an error within the preheat plugin: https://github.com/marian42/octoprint-preheat/blob/master/octoprint_preheat/__init__.py#L286. There also should be a log message which starts with `Preheat error: ` maybe also search for that.

davide-molino, 2020-10-15:
> I just tested this with a 320x320 thumbnail and everything works as expected. Could you maybe provide the file so I can test it with the exact same file?
>
> I'm thinking that this is probably a [...]

I was thinking the same... but the error shows up with various resolutions.

davide-molino, 2020-10-15:
> I just checked the API code for the preheat plugin and the 405 indicates an error within the preheat plugin: [...]

I think that this will be the solution: 
"If you have a printer that adds image data to the top of the G-Code file (Prusa Mini), the default of checking the first 1000 lines for set temperature commands might not be enough. Use the setting labeled "Max number of lines to look for preheat commands" to adjust how many lines are looked at."

at the bottom of the page https://github.com/marian42/octoprint-preheat

UnchartedBull, 2020-10-15:
Great find! Going to add this to the troubleshooting guide. Did it work for you?

davide-molino, 2020-10-15:
> Great find! Going to add this to the troubleshooting guide. Did it work for you?

Yes it works... i'm trying with 512x512 size preview and it seems to work... I've set the check for line number to 5000

UnchartedBull, 2020-10-15:
Great, also has been added to the troubleshooting guide: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#preheat-button-isnt-working-once-preview-images-are-enabled. I'm closing this here now. If the issue comes back, please reopen.

closed 2020-10-15 by UnchartedBull